ClockSelect/myevic #240: Suggestion: Show how long the current coil has been in use

NoisyFlake, 2017-01-06:
I think it would be a nice feature to activate a setting when a new coil is used that will then track the amount of days for how long the coil has been used so far (on devices that support RTC). I usually use my coils for 12-14 days so it would be nice to know when the time has come to replace it.

falconettt, 2017-01-07:
IMO:
It makes no sense to count days if you use the SS316. 
Over time the resistance increases. 10-20% growth is a good signal to change the coil.

Justin-L-Franks, 2017-01-10:
I don't really see much use for this. You can have two or more coils that the firmware will recognize because they have similar resistances. And "time used" isn't even a good metric to determine when a coil needs to be changed. Different build types "age" differently -- my Clapton coils stay "good" for 3+ times longer than my simple single-strand coils. Different types of wire, and even different gauges of the same type of wire, degrade at different rates too. Different juices will also affect a coil's lifespan, especially if a certain juice necessitates frequent cleaning (whether by dry burning or brushing / cleaning).

Coils should be changed when they go bad, determined by changes in their appearance, performance, and taste.

ClockSelect, 2017-01-12:
I agree with the two posters above.
What's more, it would only work correctly on RTC-capable boxes.

closed 2017-01-13 by ClockSelect