ClockSelect/myevic #241: Suggestion : logo/image on charging screen

popokoplo, 2017-01-09:
maybe it will more exciting if there is an image when the device was charging. i found this patch on evic vtwo mini 4.02 firmware.

thx, for make this awesome firmware :)
[image]

ClockSelect, 2017-01-12:
It would imply to have an image loader software for that purpose.
This doesn't exist at the moment.

ClockSelect, 2017-01-19:
I'll just make an empty picture in the image db you could change to whatever you want, simpler option.

ClockSelect, 2017-01-21:
You may upload a picture in the character #0xFF of the font.
If something is present, it will be used as the loading screen.

Use VTCFont or NFirmwareEditor to import an image.

closed 2017-01-24 by ClockSelect