ClockSelect/myevic #242: Please add both imren & awt batteries because its trending in Middle East, I have made a video about this firmware in youtube and its booming

1thinkdifferent, 2017-01-10:
**Box Model:**

**Coil setup:**

lojpek, 2017-01-10:
IMR and AWT are rewraps so it's not able to do profiles. All rewraps have parameteres that you can't  compare. Two accu from AWT can have totaly incomparable parameters of discharging.

ClockSelect, 2017-01-11:
@1thinkdifferent  Thanks for the video, I appreciate your support.

@lojpek That's right. Any battery that is not a Samsung, Sony or LG is in fact a Samsung, Sony or LG rewrapped with the vendor brand. And if it's not, better not buy it.
What's more, a same vendor battery model can correspond to several genuine manufacturer models over time (and even on a same period of time).
There's some ressources over the web that can help you find out what's really under the wrap; most of the time, it will be one of the already featured cells.

closed 2017-01-14 by ClockSelect