ClockSelect/myevic #245: Configuration tool

membranepotential, 2017-01-14:
Hey!

Some rumors are going around that there is a config tool for myevic in the makings.

I also hacked together a tool written in python and pyqt, I wouldn't say it is publishable yet, but I wondered if you want to release some infos to see if we can join forces? I think the config tool would be a very important step to take the custom firmware thing to the next level :)

The problem with the NFETool for me is that it only runs on windows. So I think platform independence is really necessary.

Just writing this, I had an idea: a Python tool is quite platform independent, but probably difficult to install on some platforms, especially when QT is involved. So how about a thin wrapper that runs a local HTTP server and translates to the HID interface? Than the interface could be written in javascript, the wrapper should be easy to adapt to Windows, Mac and Linux and platform independence is ensured.

I am spending way too much energy on the vaping stuff :P

Cheers

ClockSelect, 2017-01-14:
Well, with sources published, all infos are more or less public.
Just ask if you need some more details of some sort.

The plan actually is to enable the myevic configurator of the NFE, since everything is almost done already.

Ideally, such an app should be stand alone and not requires installation of third-party stuff such as python or Qt.

enricorov, 2017-01-14:
@ClockSelect We are installing a custom firmware on our boxes! I think all of us have python and qt on our devices (:

membranepotential, 2017-01-14:
Ah ok I though a separate tool will be released... Thats too bad, but then I probably will publish the python tool once the main features are implemented, so that mac and linux users have something to play with, too

Justin-L-Franks, 2017-01-15:
@enricorov "We are installing a custom firmware on our boxes! I think all of us have python and qt on our devices." That's definitely not a given, as installing this firmware is no more difficult than updating the factory firmware (just download the binary and flash it with the factory firmware update tool). A platform-agnostic solution is definitely preferable.

@membranepotential "I probably will publish the python tool once the main features are implemented, so that mac and linux users have something to play with, too." Awesome, looking forward to it. BTW, a bit off topic, but a good number of Mac users have Windows installed on their machines as well, since there are many applications which are Windows-only. On my iMac, I have a Windows 7 partition that I can either boot directly into via Bootcamp, or run as a virtual machine from within OS X using Parallels. Almost everything runs just fine with Parallels. Even most modern games only take a ~10-15% performance hit running on the Windows virtual machine compared to booting direct.

enricorov, 2017-01-15:
@Justin-L-Franks Of course, I was kidding. It's definitely more preferrable to have a tool that doesn't depend on anything else. I'd be happy to try one, no matter what I have to install first, that's all.

nbraunstein, 2017-01-23:
I would enjoy testing your app in the meantime. Can't wait for this button to start working. The NFE Toolbox, there is a button under Artic Configuration that is for myevic but I am unable to push it. Can use device monitor on 6.40 but gives me a weird readout on 6.44

ClockSelect, 2017-01-23:
We're all waiting^^
Me first, obvioulsy.

nbraunstein, 2017-01-23:
I can't wait :) this is amazing. Installed ClockSelect then Artic Fox and now back to Clock Select. Device Monitor works flawlessly on 6.40, weird values on 6.44. I am anxiously awaiting an update! I'm sure you are sick of answering but when can we expect the next update? Is there an email list or should we just subscribe to this thread to stay informed?

nbraunstein, 2017-01-23:
Is there a way to get the clock to stop showing? I've got the screen saver set to "none", saver set "off" but it keeps flashing back to the clock after about a second of pushing the fire button on my rx 2/3, worked fine until I switched to two batteries from three. Also, it's hard for me to select a new wattage, display mode...it keeps flashing, won't respond to the fire button or fire and right button to save changes. Seems to be a little bug...

nbraunstein, 2017-01-23:
Even if I put screen saver on one minute, change it to "snow" it will still show the clock almost immediately after vaping.

ClockSelect, 2017-01-23:
Interface>Clicks, disable the clock on double-click

nbraunstein, 2017-01-23:
Oh I did that, seemed to not want to save the settings. I've tried everything, even re-flashing. Seemed to just happen all of a sudden, I'll get it figured out.

ClockSelect, 2017-01-24:
The last release should work fine with the last NToolbox monitor

nbraunstein, 2017-01-24:
The configuration tool also? The button it won't let me
press?

ClockSelect, 2017-01-24:
No not yet unfortunately, it's just for the device monitor.

nbraunstein, 2017-01-24:
Can't wait, hopefully it will all be available very soon. Counting down...days I hope.   
  
  
  
 Sent from my iPhone, please excuse any syntax errors.

nbraunstein, 2017-01-29:
No updates as of yet?