ClockSelect/myevic #246: Wismec reuleaux rx300 not outputting 300w

xxnovosxx, 2017-01-17:
Sry for my bad english 

My wismec reuleaux rx300 isnt outputting the 300w when i fire the screen is showing 160w and then 250w the batteries are fully charged thx for the nice firmware

ClockSelect, 2017-01-18:
What's your setup (reststance, battery model etc) ?

xxnovosxx, 2017-01-18:
0.25 ohm and 4 LG DBHG21865 3000mAh 3,6 - 3,7V 20A

ClockSelect, 2017-01-18:
250W is what it takes to drain your batteries at their nominal drain capacity of 20A per cell, so I think it's perfectly normal (it's just above 60W/cell).
To reach to full 300W, you'd need 25A+ rated cells, like 30Q, 25R, VTCs, etc.

If you're using the generic battery setting or the HG2 setting (in the Expert>BAT menu), the box will enforce the max drain of 20A/cell, thus 250W.

Of course you can fake the box by telling it you're actually using another model of cell, but you'd get inaccurate charge readings and you'd stress your cells and lower their lifespan.

MrScaryZ, 2017-01-30:
throw some samsung 25r in or the VTC5a yesterday in bypass mode it hit 293watts burned my rda cotton to a crisp the chip surely can output the wattage

ClockSelect, 2017-01-30:
@MrScaryZ your sacrifice for the sake of science is greatly appreciated :man_scientist: 
:)

MrScaryZ, 2017-01-31:
@ClockSelect I am a great Martyr or maybe I lick kicking my own ass ;0 HAHA

ClockSelect, 2017-01-31:
I think answer has been given.

closed 2017-01-31 by ClockSelect