ClockSelect/myevic #247: question? how can i copy my config to other battery

acevalsl, 2017-01-18:
hi, how can i transfer my config from vtwo mini to other vtwo mini identical........??

thxs

**Box Model:**

**Coil setup:**

ClockSelect, 2017-01-18:
By using a tool such as [NFirmwareEditor](https://github.com/TBXin/NFirmwareEditor/releases), you can read and write the Dataflash from and to a file.

acevalsl, 2017-01-18:
i am using that tool but nothing happend.
flirt read old vtwo, and next write new vtwo with file created in firs step, but no copy nothing.
i tried nfirmwaretool and nfetool box

ClockSelect, 2017-01-18:
The two boxes should also run the very same firmware version, but I assume it's obvious.
It *should* work; I can't really help you there. I've no two identical boxes so I can't test by myself.

You could also use [Ban3's python-evic](https://github.com/Ban3/python-evic) as a command line tool if you're on Linux.

acevalsl, 2017-01-18:
all same but nothing.
i am using os x
i am going copy parameters by hand.
no problem

[image]

ClockSelect, 2017-01-18:
Nice couple :)

closed 2017-01-19 by ClockSelect