ClockSelect/myevic #248: Suggestions for preheat and power curve modes

Justin-L-Franks, 2017-01-19:
After using the firmware for a while, I thought of some suggestions to the preheat and power curve modes.

1. Add a "post-cool" setting to the preheat mode. At a configurable time, allow for a reduced power (configurable via percentage of the selected power setting) for the remainder of the firing. This would be useful for those who take very long draws, by preventing excessive heat build-up. While the power curve mode allows for this to be done, the preheat mode is easier for those who don't want to delve into the more complex power curve mode.

2. Separate the combined preheat / power curve delay time into two settings, each in their respective submenu. It doesn't make much sense to have to go to the preheat menu to set the delay time for the power curve mode.

3. Add an option for the delay times to maintain power curve settings below 100% (or the "post-cool" setting if that is added). By this I mean that during the delay time, all power curve settings above 100% would be ignored, but it would still reduce the power for any time periods set to below 100%. So if you have a 120% power setting for the first second, then 100% until 5 seconds, and 80% after 5 seconds, if you fire during the delay time, the power output would be 100% for 5 seconds, then 80% thereafter. Similarly, if you have a preheat setting of 120% and a post-cool setting of 80%, with this option enabled, during the delay time the preheat will be ignored, but power will still be reduced to 80% during the post-cool period.

This will be helpful because the delay time is there to prevent excessive heating of a coil that hasn't yet fully cooled when chain vaping. But as implemented, the entire power curve is ignored during the delay period -- even the settings below 100% -- continuing to heat the coil at full power when you want it reduced towards the end of long draws.

4. Allow additional power curve settings above 10 seconds when the long fire protection time is greater than 10 seconds. At least as it is now, whatever your last setting (9.5 - 10.0 seconds) is maintained after the 10 second mark instead of it reverting to 100%, but further control should be allowed if the long fire protection is set to more than 10 seconds.

ClockSelect, 2017-01-19:
Phew! Where to start? :)

1. Technically, a Preheat power followed by the main power followed by the Post-cool power *is* a power curve. A three steps only, I agree, but it starts to become complicated too. Editing a power curve is a bit cumbersome, but it's done once and for all. And once we'll have a real curve editor, it'll make things easier.

2. Yes, it was a shortcut. Maybe I could duplicate the delay item in the Curve menu.

3. During the Delay period, the applied power is the min(100%,setting%), so it already does what you want. It won't fire 100% if you configured a 80%; it will stick to your 80%.

4. Problem is: it takes config space, and makes the curve even more cumbersome to edit. The system should be redesigned to have (time,pwr) couples instead of one point each 0.5s. This would address both points 1. and 4., but it's a bit of work. Is further control really needed after 10s?

gpolydoros, 2017-01-19:
While you're at it, please consider adding an additional check: I've noticed than even a simple (0.1s) click of the fire button will trigger a delay, if the delay timer is set. This is potentially a problem because sometimes misfires happen or maybe you're in stealth mode and you simply want to check your battery level. In such cases, you lose the preheat function immediately - although the coil is obviously still cold. So maybe you could store a lastFireDuration var and check whether it's greater than say one second before setting the delay timer. One second is probably enough to indicate that the last click of the fire button was indeed an attempt to vape.

membranepotential, 2017-01-19:
@gpolydoros I don't think this happens for me. A short press on fire does not trigger the delay timer (Vtwo mini)

@Justin-L-Franks 
1. I think Power Curve settings should be the preferred way to configure that kind of behaviour

2. I think a separate delay setting for the power curve would make sense – or even better: indicating by menu redesign that the delay timer affects both power curves and preheat function

4. I like the idea of having (time, pwr) settings, it would make power curve editing on the device easier, but I don't think control after 10s is needed. I personally never fire longer than 10s. Maybe it could be expanded to 15s since this is the longest possible time before protection sets in anyway.

IMHO, integrating preheat and powercurve into one setting makes the interface easier, and that's always worth it right?

gpolydoros, 2017-01-19:
@membranepotential Interesting. Just checked again with my Vaponaute LPB and if click the fire button, the fire duration bar appears on the screen (registering a 0.1s puff) and the curve (C) symbol starts blinking. I suppose it could be that my fire button is of bad quality.

membranepotential, 2017-01-19:
@gpolydoros I tried again in curve mode, and can say: firing for 0.1 s with time=0.5s and delay=10s → the curve symbol is not blinking. The mystery continues...

gpolydoros, 2017-01-19:
@membranepotential Found it, time was the culprit! I had time set to OFF since I thought it was only relevant to preheat... I set it to 0.5s and now it doesn't blink until I reach the 0.5s mark! Thanks!

gpolydoros, 2017-01-19:
Maybe moving curve under preheat (as a submenu) would be better UI-wise. I think it would make it far more comprehensive for the average user that time and delay also relate to the curve. You'd have to go a bit further to reach it, though.

ClockSelect, 2017-01-19:
@gpolydoros: The time setting should not be relevant for curves; little glitch. And yes, menus should be reorganized a bit - time-consuming task.

@all: You can try [this prototype](https://www.dropbox.com/s/6878hw1w0ufei1t/myevic.bin?dl=1); I modified the curve edition to account for ten points (time,pwr) over 25 seconds. To delete a point, set the pwr value equel to a nearby point. (not parctical but I've made this quick, will improve later on).

Tell me what you think.

Justin-L-Franks, 2017-01-20:
@ClockSelect _"Technically, a Preheat power followed by the main power followed by the Post-cool power is a power curve. A three steps only, I agree, but it starts to become complicated too. Editing a power curve is a bit cumbersome, but it's done once and for all. And once we'll have a real curve editor, it'll make things easier."_ Understood. I just thought that having an easy way to set up a basic 3-part curve (preheat, normal, then reduced power) would be helpful, but a better curve editor would accomplish the same results.

_"Yes, it was a shortcut. Maybe I could duplicate the delay item in the Curve menu."_ By duplicate do you mean that if you set a delay time in the power curve mode, it would be setting the same delay time to *both* the power curve and preheat modes? I think that would be even more confusing than the existing setup. A separate setting, independent for each mode, is a better option.

_"During the Delay period, the applied power is the min(100%,setting%), so it already does what you want. It won't fire 100% if you configured a 80%; it will stick to your 80%."_ I just checked and verified that this is working exactly like I suggested. Was it always this way? I'm almost certain when the power curve mode was added, that the curve was completely disabled during the delay period.

_"Problem is: it takes config space, and makes the curve even more cumbersome to edit. The system should be redesigned to have (time,pwr) couples instead of one point each 0.5s. This would address both points 1. and 4., but it's a bit of work. Is further control really needed after 10s?"_ An array of time/power couplets would definitely be a great way to implement this. It would allow finer control of the power curve, *and* allow adjustments after the 10 second mark. Obviously, if there is a way to squeeze in more than 20 couplets that would be ideal, but even 20 time/power couplets is a better solution than the current 20 settings at fixed 0.5 second intervals. Even if you had to remove the Flappy Bird game to do so, I think it would be worth it, as gaming on a mod is rather pointless. And it would free up additional space for additional (useful) features in the future.

Thanks for taking my suggestions into consideration. Separate delay time settings for the preheat and power curve modes, along with time/power couplets for the power curve, would cover all of my suggestions.

Justin-L-Franks, 2017-01-20:
@gpolydoros _"While you're at it, please consider adding an additional check: I've noticed than even a simple (0.1s) click of the fire button will trigger a delay, if the delay timer is set."_ I *knew* there was something I forgot! I had already noticed that behavior myself.

@ClockSelect To address this, you could add in the same check that is performed when determining when the puff counter should be updated. If you fire for 0.5 seconds or less, the puff counter isn't incremented -- only firings of 0.6 seconds or longer are counted. I think that would be sufficient to prevent inadvertent fire button presses from activating the delay timer.

ClockSelect, 2017-01-20:
@Justin-L-Franks did you [test ](https://www.dropbox.com/s/6878hw1w0ufei1t/myevic.bin?dl=1)?

gpolydoros, 2017-01-20:
@ClockSelect I see! Thanks for an awesome, open source firmware though! I've been using it as a daily driver for all of my mods for a few months now and I've never really thanked you enough. If you ever decide to add a PayPal donation button, let me know and I'll happily donate. :)

On a side note, do you think it would be feasible to build a similar editor for non-linear TCR (essentially creating a TFR curve) in the future? I know it wouldn't be simple; I'm just asking out of curiosity, since this feature is currently available on the DNA, and with good reason.

ClockSelect, 2017-01-21:
@Justin-L-Franks a seperate delay does not seem necessary to me. Using the curve or the preheat is basically the same thing; preheating is a basic two-step curve. When I'll redo the menu layout, it will become more clear.

The last release should already cover most of your points.

ClockSelect, 2017-01-26:
@Justin-L-Franks what do you think of the improvements?

Justin-L-Franks, 2017-01-26:
@ClockSelect I'm still testing; I've been busy the past several days and haven't had much time to really dig in and check & test everything. At first glance it definitely seems to be on the right track.

Justin-L-Franks, 2017-02-01:
Found a bug. The power curve is maintained between set points, but after the last set point, it reverts to 100%.

To reproduce:

1. Insert a point at 0.0 seconds / 110%.
2. Insert another point at 25.0 seconds / 109% (it needs to be different than 110% in order to "take").
3. Fire, and the power level will remain steady at 110%.
4. Delete the second point, leaving only the 0.0 seconds / 110% one.
5. Fire, and the power level will remain steady at 100%.

The same thing happens when more than two points are set -- as soon as the last point is passed, the power immediately reverts to 100% instead of maintaining what you put in the last point.

The only exception is when the maximum allowed number (10) of points is set. Whatever you set as the final point **is** maintained throughout the rest of the firing.

For instance:

1. Insert the following points:

0.0 s / 90%
0.5 s / 91%
1.0 s / 90%
1.5 s / 91%
2.0 s / 90%
2.5 s / 91%
3.0 s / 90%
3.5 s / 91%
4.0 s / 90%
4.5 s / 91%

2. Fire, and the power level will stay at 91% past the 5 second mark.

3. Delete the last point, leaving the first nine in place.

4. Fire, and the power level will revert back to 100% past the 5 second mark.

ClockSelect, 2017-02-01:
I see

ClockSelect, 2017-02-01:
It should behave a little better :)
Thank you for your thorough testing

Justin-L-Franks, 2017-02-01:
Confirmed. The last point on the power curve is maintained throughout the rest of a firing, regardless of how many points are set.

Thanks for fixing this so quickly!

Justin-L-Franks, 2017-03-10:
After using the new power curve editor for a while, I find that I'm running up against the limit of 10 time/power couplets. Would it be possible to allow for 15 - 20? Even just one or two more would be great, as the power curve I want to use is a 115% preheat that smoothly drops down to 100%, followed by a "post-cool" with a smooth decrease down to 85%:

- 0.0 s 115%
- 1.0 s 110%
- 1.5 s 106%
- 2.0 s 103%
- 2.5 s 101%
- 3.0 s 100%
- 4.0 s  99%
- 4.5 s  97%
- 5.0 s  94%
- 5.5 s  90%
- 6.0 s  85%

This requires 11 time/power couplets, so in order to use only 10 entries, I've been bouncing back and forth between a 110% - 85% curve and a 115% - 90% curve.

If the long fire protection needs to be dropped down to 15 seconds in order to accommodate additional power curve entries, I think that would be worth it. I take much longer draws than any other vaper I met, and a maximum cutoff of 15 seconds wouldn't affect me since I almost never go beyond 12 - 13 seconds.

Thanks so much!
-Justin Franks

360up, 2017-03-31:
> The system should be redesigned to have (time,pwr) couples instead of one point each 0.5s.
Yes please, for quick puffs 0.5s intervals are too long. It would be great to be able to adjust the time too.