ClockSelect/myevic #249: Flip up or down button?

AAI688, 2017-01-19:
Hi, using egrip 2 now and its button layout is vertical so + is at the top and - is at the bottom. somehow we are used to + and - buttons being horizontal that navigating on other mods is not too hard but on egrip 2 i find it confusing as + is down and - is up (opposite of its layout). can we have a setting where we can flip that on certain mods like egrip 2?

thanks.

**Egrip 2**

ClockSelect, 2017-01-19:
I see...
It wouldn't be easy.
Thay've been a bit stupid to do that in the first place. I guess it's for gaining a few millimeters in width.

What's really hard to use with this setting?

AAI688, 2017-01-19:
as i said, the up button is now down and down is now up in menus which navigating is very confusing.. i was thinking that if you cant add a setting for this you could do an overall change in the buttons as the other way round but that would be selfish of me for you to do that to everyone..

ClockSelect, 2017-01-21:
@AAI688 Test [This](https://www.dropbox.com/s/6878hw1w0ufei1t/myevic.bin?dl=1), tell me if it fits better.

AAI688, 2017-01-21:
yup this one works well!!

ClockSelect, 2017-01-26:
:)

closed 2017-01-26 by ClockSelect