ClockSelect/myevic #251: RX 300 weak battery message-- so the batteries are ready 3.64 and still get weak battery on fire

MrScaryZ, 2017-01-19:
Batteries HE4

**Box Model:** RX 300

**Coil setup:** .16 ohm dual coil SS .304 running wattage

MicroSur, 2017-01-20:
Wattage is necessary to be said.

MrScaryZ, 2017-01-20:
I just tested so at 3.81v on the battery at 50 watts or above it claims weak batteries they are all HE4'S
same ohms of coil is .16

ClockSelect, 2017-01-21:
That's strange. How old are your HE4's?

docdeath, 2017-01-21:
i own rx300 to and have run it down to 15% on battery with 3,16 volts and not had this on both 25r and 30q  coil i use .050 at 35 watts  at 15% i get weak batteries   cant test other coils as don't have any

MrScaryZ, 2017-01-21:
The HE4's are brand new I also tried some HG2'S same issue I am wondering if its a amp draw with that .16 ohm dual coil

ClockSelect, 2017-01-22:
This could be amp draw if you were running around 250W or more, but not 50W.
Things work fine with stock firmware?

Something you can try: clean the contacts of the box (and cells), it may just come form there.

MrScaryZ, 2017-01-22:
Yep I sprayed it with some electronic cleaner and am cycing the batteries
on the OPUS  but it is still doing it and now its doing it with a .463 ohm
coil just a p rebuilt aspire cleito.  This does not happen with the stock
RX 300 firmware and its still happening as soon as I go to 50watts

ClockSelect, 2017-01-24:
I'm a bit stuck here since I have no RX300 by hand to test.
I'll try to find one, but it may take a few days.

I really can't see why you'd have an issue at 50W.

MrScaryZ, 2017-01-24:
I concur it makes little since it is not happening on my RX 2/3 or the RX 200s with this firmware just the RX 300 it has to be some variable with batteries being 4 vs 2  vs 3 if there is anything else you would like me to try do I have a bit of time here and there to assist

ClockSelect, 2017-01-24:
Found the bug.

MrScaryZ, 2017-01-26:
@ClockSelect awesome

closed 2017-01-26 by ClockSelect

MrScaryZ, 2017-01-26:
@ClockSelect i will load the newest build today and let you know how she behaves

MrScaryZ, 2017-01-27:
@ClockSelect loaded latest firmware all appears fine now thanks for the great work