ClockSelect/myevic #252: ML/KJ for TFV8

Gremdude, 2017-01-20:
Hi there!!

The default ML/KJ value for this firmware is 360. Per the readme, this is a good value for a standard coil at 30w. Well, I am used a TFV8 with the Octocoil so obviously, this value is not gonna do it! What value would you recommend for my setup? I am using 900 right now as this is about 3 times the standard value and seeing how the accuracy is. Is there any way to set up the firmware so it adjusts this ML/KJ automatically based on the ohms load and wattage used? Thanks for your time and the continued hard work you are doing to make these mods much better than when they left the factory!

nikhill666, 2017-01-20:
I have mine set at 300 on my RDTA build, dual coil firing at 45w - the tank holds 4ml and by the time I've drained it - the ml used is almost spot on. 360 was to high for me. 🙂

gpolydoros, 2017-01-20:
I have mine set to 250. The process of making it "accurate" is simple, you just reset the counter with the tank full and then adjust the ml/kJ ratio when it's gone empty, so that it reads the exact capacity of your atty. But, honestly, I would be surprised if the firmware could adjust this ratio automatically, since it doesn't only depend on coil and wattage, but also on juice density and probably other factors of such nature that are impossible for the firmware to account for.

ClockSelect, 2017-01-21:
Liquid consumption is really a matter of Joules - that is, Watts by seconds. Maybe some other factors play around, but the Joules are the only thing the box can compute by itself.
900 ml/KJ seems a bit high to me; it would mean you vaporize three times as many liquid as me with the same power.
The method described by @gpolydoros to setup the value should give you a good result.

closed 2017-01-26 by ClockSelect