ClockSelect/myevic #253: Resistance reading and shunt resistance setting on Cuboid 1.01

Gremdude, 2017-01-20:
Hi again!

My Cuboid is reading the resistances WAAAAY off. My .15 ohm Octocoil in my TFV8  is reading .20 which is kind of ridiculous. There are variances in every mod when they read resistances but normally we are talking .01-.02. NEVER .05. I know this is a wattage coil so it doesn't make a huge difference but in temp control it is a killer. Your firmware has an ohms adjustment feature (thanks so much for this BTW--all mods should have this!) but it would be nice if there was a way to adjust the mod so this variance could be compensated automatically. I notice that you have a "SHR" value which can be adjusted in the Expert menu. When I adjust this Shunt resistance, it does bring the numbers down to a more reasonable value but for some reason, it makes the entire vaping experience go out of wack! For example, I have a Nickel Coil on my TFV4. The coil was reading very high at .15ohms instead of the correct .11-.12 rating of the coil. I adjusted the SHR down to 80 and read the resistance. The resistance was now reading correctly at .12 ohms but it was firing WAAAYYYY hot for some reason. On my TFV8, I set the shunt resistance to 80 to see if I could correct the variance. Now the resistance was reading the correct .15 but when I fire, it is really cool at 90 watts. Feels like 60 watts or less!! Am I misunderstanding the purpose of the SHR? It seems to be doing what I want it to do in terms of making the resistance read correctly but the result it has in terms of vape quality is the OPPOSITE of what I am expecting. Thank you for your time and continued hard work on this project!

ClockSelect, 2017-01-21:
The shunt resistance is used to read the current flowing through the atomizer. It's mounted in serial between the ato and the power supply. So it serves two purposes: reading the atomizer resistance, and reading the effective power.
If you modify the shunt for resistance reading, it's also modified for power reading. If the shunt value was really your problem, you'd have a power issue too and modifying the shunt would solve both problems at once.
But since solving the resistance problem creates a power issue, it means that your shunt value must be good in the first place. Your Cuboid is bad at reading atomizer rez for some other unkonwn reason.
Adjusting the rez via the coil management seems the way to go.

Eddieb1213, 2017-01-24:
Have the same problem with my Cuboid. Was always reading my resistance correctly or maybe ,01-.02 off but since i installed myevic firmware it is off around ..03-.05 on prebuilt coils [dont build my own coils so ohms are always known]. Everything else works flawlessly and love the firmware but was wondering if there is a adjustment or fix for it

ClockSelect, 2017-01-24:
Could you try [this version](https://www.dropbox.com/s/6878hw1w0ufei1t/myevic.bin?dl=1) and tell me if you see an improvement?

Gremdude, 2017-02-05:
Sorry but that version did not help much. It made the variance less but to be honest, even though the displayed resistances are way off, the mod is doing an EXCELLENT job of temp control so in my opinion, it is not really an issue. For example, right now I am vaping the Uwell Crown with the notch coils from Coilart and it is reading .325 ohms rather than the correct .3 ohms. Normally I would be concerned when the resistances are reading so high as it would lead to an extremely hot vape but this is not happening, The device is able to work properly using these displayed readings. That being said, you could close this thread if you want.