ClockSelect/myevic #254: RX300 report

Pilou59, 2017-01-22:
**Box Model:**
Reuleaux RX300
**Battery Model**
EFEST Purple 35A 3000mAh

Hi everyone,

First, sorry for my bad english...
I've got a problem on my Reuleaux, 1 battery dischage so quickly before other. 
I've shutdown the box yesterday, with 4 full batteries. Today, one of the has 1.6V, the other one have got 3.3V, 3.5V and 3.6V. My charger is a efest LUC V4.
I had no problem with the original firmware. I'll flash it back and follow this thread.

Do not hesitate to correct my English

ClockSelect, 2017-01-23:
That's weird.
The 1.6V seems just impossible.
Does it makes a difference with Expert>LSL ON or OFF?

Pilou59, 2017-01-23:
I'll try that

lojpek, 2017-01-24:
**EFEST Purple 35A 3000mAh**a
It's rewrap and others aku's in your box can have others parameters. IT IS REWRAP and you people must understand that always can have other parameters for discharge and others parameters for mAH and max Ampers.

ClockSelect, 2017-01-24:
A rewrapping brand can indeed change the model of the rewrapped cell through time - so you may have cells with different characteristics under the same wrap.

But it's unlikely to happen under a same batch of cells; if you've bought them together at the same time, you should have uniform characteristics.

lojpek, 2017-01-25:
Rewraps are not same as original, even when under wrap you have original HG2 or something else. Rewraps like EFEST are "building" from original battery that was rejected in production line from some reason. There are buying from SAMSUNG or LG and wrapping as a EFEST or something other. I say no to rewraps for your safety.

ClockSelect, 2017-01-26:
@lojpek You're perfectly right. But few people know that Samsung & LG are the only two real cell manufacturers.
You're a bit hard when you say "rejected"; rejected cells are really rejected. Rewrapped cells are just "not garde A". It makes quality vary from cell to cell, but not at the point the OP experiment.

Anyway, never buy something not from Samsung or LG.

MrScaryZ, 2017-01-26:
@ClockSelect you are saying Sony does not make batteries?

ClockSelect, 2017-01-26:
@MarkyAD Yes.
Sony contracts some factories to manufacture batteries given some specifications for some specific needs. Those batteries are not intented to be sold to anyone beyond Sony's target customers, but factories, sometimes, produce just a bit too much batteries. As an example, Sony ended ordering VTC4 production in... 2005. Just funny.
None of the VTC4,5,6 or whatever you can have in your mod has never been quality controlled by Sony. And you won't ever know wich factory it came from.

MrScaryZ, 2017-01-26:
@MarkyAD as with most products they contract out their design makes sense

lojpek, 2017-01-26:
@ClockSelect you are right but in my opinion those who are not grade A are rejected. It can be several spec like not having max Amp or more heating when used. But rewraps has 35A and 3500mAh and under wrap you have max 20A and 2600mAh. It is our health and I never accept roulette with batteries. We have electronic mod for vaping but it's only electronic. Everyone who are using boxes or mechanical mods must know that is our health and it's not game - maybe explode or maybe not.

Pilou59, 2017-02-19:
So, I've tested it.

Batteries level :
1 - 3.65
2 - 3.67
3 - 3.66
4 - 3.62
It's 11:45, batteries at charge. They're full at 13:00 at 1A
So, I've flashed the myevic firmware in my RX300
LSL is ON.
I shutdown the box at 13:20. Wait and see...

Pilou59, 2017-02-20:
So, today, I have this levels :
1 - 3.83
2 - 3.97
3 - 3.46
4 - 3.91...

I tried to install arcticfox. X32 is ON and LSL off.
How can arcticfox have x32 on, wherever it's off on myevic. Is there a x32 crystal on RX300?

fOmey, 2017-02-21:
Have you actually checked the voltages with a multi-meter or voltage checker ? The firmware could be reporting incorrect battery voltages.

Pilou59, 2017-02-21:
I have no probleme with arcticfox firmware.
I check batteries with external charger. The voltage shown on charger is the same (0.02V max of difference) that in the box (even full or discharged)

fOmey, 2017-02-21:
OK, I want to make sure I understand what your trying to explain.

As you use your mod and you discharge your batteries, they are being discharged unbalanced ? If your box reads 1 - 3.83, 2 - 3.97, 3 - 3.46, 4 - 3.91 and you place the batteries into your charger will your charger also report the same voltage readings: 1 - 3.83, 2 - 3.97, 3 - 3.46, 4 - 3.91 ?

I have noticed that my mod reports my batteries are being discharged unbalanced, however when I check the voltage on my batteries they are not. The mod is reporting the wrong voltages as they discharge.

Pilou59, 2017-02-24:
> As you use your mod and you discharge your batteries, they are being discharged unbalanced ? If your box reads 1 - 3.83, 2 - 3.97, 3 - 3.46, 4 - 3.91 and you place the batteries into your charger will your charger also report the same voltage readings: 1 - 3.83, 2 - 3.97, 3 - 3.46, 4 - 3.91 ?

It's exactly that !

ClockSelect, 2017-08-30:
It should be fixed now.

closed 2017-08-30 by ClockSelect