ClockSelect/myevic #256: myevic.bin v3.30 Bild 170124 - screensaver 'splash'?

BigBadBlackWolf, 2017-01-25:
Hi,
myevic.bin V3.30 build 170124. What is screensaver 'Splash'? By my box vtc mini nothing happen!

ClockSelect, 2017-01-25:
You may use a custom 64x128 picture to replace the character 0xFF of the font.
If you do so, it'll be displayed as splash screen (loading screen at startup), and you also can choose it as screen saver.

By default, this pic is black 'cause I don't have a cool loading pic screen yet for myevic.

BigBadBlackWolf, 2017-01-28:
Thanks for info. I have create a nice (standard)pic for character 0xFF. If you want, you can insert it in an update for all user.

ClockSelect, 2017-01-30:
@BigBadBlackWolf Sure I'd welcome pics suggestions, since I'm not really an artist myself^^

BigBadBlackWolf, 2017-02-01:
The readme.md is too old. It would have to be adapted to the current state.

ClockSelect, 2017-02-02:
Indeed, this item is not documented yet.

BigBadBlackWolf, 2017-02-02:
It would also be useful to specify the myevic version number and builddate so that the users know which version / build this readme.md refers to.

BigBadBlackWolf, 2017-02-11:
Too bad, but despite my advice, the Readme.md has unfortunately still not adapted to the current version of myevic V3.30 build 170201.