ClockSelect/myevic #258: Device Monitor freezing with UCH setting switched off

nbraunstein, 2017-01-27:
**Box Model:** RX 2/3

**Coil setup:** 316L 

Getting a bunch of odd USB errors on Windows 10 for the Device Monitor when I switched off UCH to avoid my computer charging my cells when I am trying to monitor my mod. Connects and graphs data for a second or two and loses connection again, wont connect, froze the application.

ClockSelect, 2017-01-30:
The only box I have to test this function is a VTC Dual, unfortunately.
Does the monitor still work fine with UCH ON ?

TBXin, 2017-02-01:
@ClockSelect device is going to sleep after 3 min delay and when it sleeps it won't send monitoring data.
UCH can affect this behavior directly - when device is charging it don't sleep.

nbraunstein, 2017-02-06:
Yes!! @TBXin