ClockSelect/myevic #260: Idea for clock - Specifically on the RX series - Hardware related

lostvoid, 2017-01-30:
As the RX series does not have the crystal for the RTC, nor does any model have a battery backup for time it would be advantageous to see if integrating a battery backup and crystal would be feasible.

Does anyone have any info on the PCB or a schematic of the board for the 200s, 2/3, or 300?

If anyone has schematics for the boards, has already done a similar mod, or has any other info that I could use to create plans for any additional 3D printed parts, additional PCB layouts, ect. I would be more than happy to publish the plans once I know that the mod is feasible and that an initial design was created.

ClockSelect, 2017-01-31:
There are some pics here and there of the PCB of some boxes.

On some of them, you may have an available slot to solder an X32 crystal, like on [the VTC-Mini PCB](https://www.dropbox.com/s/9vj8e9w8wvrtgfn/20151111_113228.jpg?dl=1)  (see the "Y2" empty slot, south-west of the MCU).

On other boxes, I don't know; you'll have to disassemble it and see if there's some space available to solder and power a crystal between PF.0 and PF.1 pins.

I've never heard of someone attempting this.

lostvoid, 2017-01-31:
Thanks for the info. Based on the pinout it should be a straightforward process to solder in the oscillator via a couple mod (bodge) wires and heat shrink and place it into an empty crevice within the casing. I know there isn't much room anywhere for the parts if there isn't solder pads on the board, but it will squeeze in somewhere.

Based on the datasheet for that chip located at http://www.nuvoton.com/resource-files/DS_M451_Series_EN_Rev2.05.pdf on page 87 I'm seeing that pin 10 for VBatt is the contact for the RTC battery backup. I'm thinking this would be suitable for a 3V lithium button cell, however, this will take more research digging through the datasheets. If this is the case 3D printing a new battery tray with an indention for a CR2016 or similar cell would be the easiest way to fit it in.

Do you know if all of the mod variants supported all use the 48 pin LQFP package?

I'm going to order some crystals tonight. I've got a RX200, RX2/3, and an RX300 on the way. Either it will work or the magic smoke will escape. Regardless, I'll let you know how it goes.

ClockSelect, 2017-01-31:
The VBAT pin is already supplied by battery power on all boxes except VTwo, AIO and Dual (RTC boxes), on which it's supplied by a mini LiPo pouch to backup RTC settings.

The problem if you try to supply this pin with an external battery (LiPo or button cell) is that you'll mess with the regular battery readings. All boxes except those three uses the VBAT pin to read battery voltage for one of their batteries.
If you try to connect a backup cell in parallel on this pin, you'll have to keep its voltage below 1.5V and protect it from the main battery by a diode, a FET or something.

ClockSelect, 2017-01-31:
More info: on all of those boxes, one of the cell is connected to VBAT through a 1/2 voltage divider. If there's more than one cell, others are connected to others ADC pins of the MCU through some voltage bridges.

lostvoid, 2017-01-31:
Good info to know. I may be overthinking the backup battery. Using a buck converter ran in parallel from the battery line into a capacitor might be easier in the long run. Usually battery swaps only take a few seconds so the cap could keep voltage going without problems and no backup batteries to change. The current draw to keep the cap charged should be negligible so battery imbalance wouldn't be a big concern.

ClockSelect, 2017-01-31:
I'm curious to see if it'll work. (Well, there's necessarily a way to make it work.)
Don't forget to capture your work on video^^

enricorov, 2017-02-04:
@lostvoid Why the buck converter? Can't you just add a capacitor on the MCU input pin?

lostvoid, 2017-02-06:
@enricorov I actually thought about that later on. I was running on no sleep when I said that. Right now I'm waiting on the crystals. Once those are in I'll update this with how it went.

donicrosby, 2017-02-08:
My head immediately went to just storing the clock time in some memory. Like @lostvoid said it only takes a small amount of time to switch out batteries. Unless every single bit of memory is used it would only take a very small amount of space to store it. It should be hard afterward to just reset the seconds counter in settings if precision is needed/wanted.

lostvoid, 2017-02-27:
i've been playing with the oscillator crystals that literally came in on a slow boat from China for a couple days. The problem I see is the pin pitch on the IC. Unless you have a decked out electronics bench you aren't going to be able to get the crystal soldered to the correct pins. There aren't traces going to those pins on the RX series either that I was able to track out on the 2/3 or the S. I guess I'm going to call the idea dead for now.

For full disclosure I haven't opened my 300 yet to look at the board. Since it's new I've been a bit protective of it and don't want to accidentally let the magic smoke out.

enricorov, 2017-02-27:
@lostvoid Thanks for the update. Unless somebody has a full schematics of the boards, we'll have to call the idea off. 

As for your RX300, don't turn it on. Take it apart!

mist, 2017-08-23:
I got an used RX 2/3 yesterday (my first box) and opened the case today but didn't pull out the PCB cause I don't want to destroy the heat-sink. But I thought about the exact same stuff and am said there are no traces.
But as @donicrosby wrote why isn't the time stored in the normal memory? Wouldn't it be possible to just store current time and date every about 10 seconds so that the clock doesn't have to be reset after every battery swap? I don't need the time to be perfect (though crystal and backup cap would be nice).

enricorov, 2017-08-24:
It's definitely feasible. Keep in mind that EEPROM memory can be rewritten a finite number of times, so writing a piece of information every 10 seconds may add up.

mist, 2017-08-24:
Ah well, didn't think about that.
But I digged through the Datasheet (Figure 6.2-7 M451 Power Distribution Diagram) and realised that `RTC & 80 bytes backup register` are powered by `V_BAT`, so if `V_BAT `is used as measuring one of the 18650s voltage it is exposed and a cap could be added easily to provide enough backup power when changing 18650s to hold the current time in the backup registers.
Am I seing this correclty? (Sorry, I'm not experianced in uC at all)

enricorov, 2017-08-24:
Not that it would matter, just something worth thinking about. Increasing the delta between backups to, say, a minute or two would solve the problem. 

Regarding the feature you're talking about, you're absolutely right, I had a look at the figure you mentioned and I don't see why one couldn't hack a capacitor in there. Size it appropriately (i.e. consider the capacity you need to provide sufficient voltage for the time it takes to swap the batteries), and you should be good to go.