ClockSelect/myevic #261: Reverting to original FW

dcorley001, 2017-01-30:
I have tried to reload the original fw on two devices without result. Myevic has performed beyond expectation on several devices but on several others with the same tank/coil and configuration in myevic, poorly. For the time being on those devices I'd like to revert back to the original FW but my attempts have failed. This has been on a Cuboid 150 and a eLeaf iPower 80w/5000mah.

ClockSelect, 2017-01-31:
If there's an eleaf iPower, the firmware cannot be myevic since it won't run (even update) on this device.
I can't help you on any eleaf device.

For the Cuboid, or any compatible device, the procedure to revert to factory firmware is just as simple as following the firmware upgrade procedure on the manufacturer's website.
(and even if you did something bad with your eleaf, it should also work).

dcorley001, 2017-01-31:
It's the iPower 80w. I've reinstalled the mfg's firmware on both but neither are reverting to original. After the process, firmware updates from myevic still remains. Love what the update has done for the eLeaf  or iStick Pico and the Wismec RX200s. They both rock!.

dcorley001, 2017-01-31:
Just retried the update on the Cuboid and the iPower 80w. It took this time.

closed 2017-01-31 by ClockSelect