ClockSelect/myevic #262: Feature Request: Time to full charge

anzejarni, 2017-01-31:
Is there a way to calculate time to full charge of the battery when connected to USB charger?

ClockSelect, 2017-01-31:
It may be possible to estimate something on some multi-cell boxes, but I'm afraid it'd be very inaccurate.
This information depends on some factors like the USB source power and the cell brand and model.
The problem is not hard *per se*, but I'm afraid it'll be too hard to reach an acceptable reliability.

I don't want people opening issues telling me I predicted 3:30 and that at 3:40 there were still 10% missing and it's a bug and please fix it.

enricorov, 2017-02-02:
@ClockSelect I love you, man.

anzejarni, 2017-02-02:
@ClockSelect, thanks for considering the option. I guess this feature will be interesting for single cell mods. Most multicell manufacturers encourage external charger for charging batteries AFAIK.