ClockSelect/myevic #264: New Management Tool

clanzotti, 2017-02-01:
Hi all,

I'm writing a new management tool for our preferred boxes. It is written in C and do not depend on third part library (It uses my own system, see here if you are interested: [dynalib demo](https://www.youtube.com/watch?v=Uzio2zztK-E) and here [dynalib](http://www.dynadream.com/ddweb/index.php/Technology). It is a little old demo and of course a lots changed since then) . It is also multiplatform and uses HID to access/manipulate device's data.

You can see a little preview of it here: [evicManager](https://www.youtube.com/watch?v=cKwF42X4cc8&feature=youtu.be)

The interface is very poor at the moment as I'm still trying to figure out how to get the HID interface working.

As you can see I'm at the data grabbing stage and so far by looking (and sometime changing) the myevic source code I was able to understand how the data are stored and how to get them. 

But I found some problems and I'm asking the dev (@clockselect) some questions:

1) I found that the HID_CMD_MONITORING report 'strange' data (As you can see in the video above) that do not match the ones I read using the vcom interface. Looking at the code I found that the returned data are treated somewhat different (maybe for a particular reason). It is incomplete or they are meant for some other usage?

2) I can't get the HID_CMD_SETPARAMS to work (that is the most important part for the tool). The command run through the system well but reaching the hidGetOutReport() function, at the HID_CMD_SETPARAMS case, the hidDFData do not match the data I passed (bytes are all 0). I write to the HID by using the correct offset and size (respectively sizeof(uint8_t) (to offset from the checksum bytes) and DATAFLASH_PARAMS_SIZE) along with the new parameters data. I do this by sending first the command and then write the data. Any hints?

Thank you for the great work on myevic!

**Box Model:**

eVic VTC Mini

**Coil setup:**

ClockSelect, 2017-02-02:
Hey, that's good news :)

1. The VCOM monitoring is deprecated and should not be used any more. The HID_CMD_GETMONDATA is the new way to do. The monitoring software issues this command at the frequency it wants (100Hz for the NToolbox).

2. Hard to help you there. The Arg1 of the HID command should be zero, Arg2 should be 2048, and the data is 2048 bytes starting with 4 checksum bytes, 252 parameters bytes, and other stuff. I can't tell what's wrong in your case.

TBXin, 2017-02-02:
@ClockSelect 10hz ^^

ClockSelect, 2017-02-02:
@TBXin oh... only? ^^

clanzotti, 2017-02-02:
Thank you @ClockSelect, it was all my fault as I forgot to add that byte for the report number. Solved it and now I'm able to write parameters without problems.

TBXin, 2017-02-02:
@clanzotti i think you should use HID_CMD_GET/SET PROFILE instead of direct dataflash access.

clanzotti, 2017-02-02:
Yes @TBXin thanks, I figured it out when I stopped on that commands to test it. :-)

ClockSelect, 2017-02-06:
Yes, profile 0 is the usual parameters.