ClockSelect/myevic #265: Not working report: RX 200 black screen after flashing

sent1n3l, 2017-02-02:
**Box Model:**
Wismec Releaux rx200

Firstly thank you for your great work! 
When i try to install myevic.bin with FWUpdater v1.1.7 i get "Firmware product verification failed" error message. I entered expert mode, clicked on LDROM mode then tried to flash myevic.bin again and it worked but the rx200 never opens (black screen).
Note: In black screen computer could find the rx200 and i flashed it again to the previous working firmware.
It looks like it's not compatible. I thought rx200 is supported? (Arcticfox firmware is working perfectly).
Thank You for your time and great work

ClockSelect, 2017-02-03:
When did you got your rx200? Maybe it's a knew hardware version.
myevic Firmware does not install on a hardware it's not sure of its compatibility.

ClockSelect, 2017-02-03:
More: do we speack of a rx200 or an rx200S?
myevic is not yet compatible with the rx200 (smal screen)

sent1n3l, 2017-02-03:
It's an Rx 200. Too bad i thought it was working when i saw rx 2/3 compatibility. I hope in the future to be supported! Thanks for your fast reply and awesome job you are doing!

closed 2017-02-03 by sent1n3l