ClockSelect/myevic #269: RX200s and 25R batt usb charging issue

userdsp, 2017-02-05:
One of my batteries was droped to 1.0 volts while mod was charged over usb!!! The wast one under the door.

TBXin, 2017-02-05:
Check your battery with voltmeter first.

userdsp, 2017-02-05:
That was the first thing I did. The volmeter also showed 1 volt.

ClockSelect, 2017-02-06:
What makes me suspicious is that if the firmware had such an issue I'd already have tons of issues opened on that subject since we have plenty of RX200S users.

Is the issue reproducible (without letting it go down to 1v - it'd kill the battery)? Do you actually see an unbabalance growing on the charge screen while charging via USB? And does the box work fine on this regard with the stock firmware?

userdsp, 2017-02-08:
This issue was happened after flashing your previous firmware. I don't have the same  problem with the last one. All batteries had 3.5 volts before I start charge them. I hope this issue is stick to the previous firmware version!

closed 2017-02-09 by userdsp