ClockSelect/myevic #271: NFE compatibility

Gremdude, 2017-02-05:
Good morning!

As always, thank you so much for the work you are doing. I am having an issue with the NFE monitor when attached to my Cubis with the latest Myevic firmware. It is giving some wacky readings as can be seen in this screenshot. 1c temp? 3600 watts? temp set 57,000F. I think that is the temperature of the surface of the sun. Pretty sure I would never better batteries to achieve that! LOL. Your assistance is appreciated!
[image]

TBXin, 2017-02-05:
@Gremdude, you are using old version of the NFE Toolbox. You need an update ^^ https://github.com/TBXin/NFirmwareEditor/releases/

Deville505, 2017-02-05:
When I tested the myEvic Firmware several times during the last 3 weeks on my Wismec RX 2/3 I had pretty much the same effect using the most recent versions of myEvic and of the NTFirmwareEditor/ Monitor, while at the same time the monitor worked fine with the respective ArcticFox firmware - it didn't seem to be a problem of outdated software (/ firmware).
The problem looked very similar, Voltage, Wattage and especially Temps were totally off in the several-(ten)thousands...

ClockSelect, 2017-02-06:
Compatibility should be restored since commit  f1b13a9774e693aa0c1b5292d7763a327f3bf427 and version 6.4.4 and the NToolbox.

nbraunstein, 2017-02-06:
Is the Comfiguration Tool coming anytime soon? @TBXin @ClockSelect

Deville505, 2017-02-06:
I'm somewhat reluctant to change firmware to check out right now (currently I'm on ArcticFox) but I'm pretty sure last time I checked I used the most recent commit of Feb 1st with NFE Toolbox 1.5 (though I'd rather bet only non-vital parts on it).
If necessary to help finding a possible glitch/ bug, I'd be willing to change to the most recent myEvic again soon and check again with the most recent NFE Toolbox. Apart from maybe doing some screenshots is there anything I could provide and/ or pay attention to in respect of myEvic settings?

nbraunstein, 2017-02-06:
@deville505 thank you :) appreciate your effort.

Deville505, 2017-02-06:
@nbraunstein I'm not sure but there seems to be a misunderstanding: my previous posting referred to the issue of the OP (the NFE Toolbox Monitor seemingly having some problems with the myEvic software). I don't work on nor have any info about any Configuration Tool for myEvic - sorry if my post was misleading you.