ClockSelect/myevic #272: Will the Joyetech Primo be supported?

ozomax1997, 2017-02-05:
I've just purchased a joyetech primo and was wondering if the firmware will be released for it, I've had this firmware on every one of my mods that has been able to have it installed and would love to have it on my primo

Box Model: Joyetech Primo

ozomax1997, 2017-02-05:
Keep up the good work!

ClockSelect, 2017-02-06:
It should be, when I'll have access to the firmware.

MicroSur, 2017-02-14:
FW is available...

hibby50, 2017-02-15:
Very happy to hear you plan on supporting it, I absolutely love your firmware on my vtc mini. Thanks so much for taking the time to develop this

Janishewskiiy, 2017-02-17:
@ClockSelect 
Firmware for Evic Primo is already available on the website JoyeTech
We look forward to his support

GravityRoller, 2017-04-17:
Is there any place user can opt in for notifications when Firmwares get posted?

BigBadBlackWolf, 2017-04-18:
ClockSelect is probably no longer available for the further development /
improvement - why ever. MiroSur has apparently accepted the case. For more
information see:
https://github.com/MicroSur/myevic/find/master?q=
Last current version of myevic is 3.30 from 09-04-2017. With Flappy Bird
and Tetris.