ClockSelect/myevic #274: NFE Toolbox

dabugen, 2017-02-11:
Hi:)

I am wondering how I can use the "myEvic Configuration" entry within the NToolbox.exe? I just can´t click it, nothing happens. If I use the Arctic Fox firmware and then the "ArcticFox Configuration" entry within NToolbox.exe, this works just fine. How can I make that work for MyEvic too?

Thank you:)

BigBadBlackWolf, 2017-02-11:
Myevic is not yet supported.

dabugen, 2017-02-11:
Ah OK, that explains it very well, thank you.

BigBadBlackWolf, 2017-02-11:
My suggestion would be to insert the following in the language packets
(here for English):

// NToolbox Window
Toolbox.MainWindow.ArcticFoxConfigurationButton=ArcticFox Configuration
Toolbox.MainWindow.MyEvicConfigurationButton=myEvic Configuration\n(not
supported yet)

My suggestion would be to insert the following in the language packets
(here for German):

// NToolbox Window
Toolbox.MainWindow.ArcticFoxConfigurationButton=ArcticFox konfigurieren
Toolbox.MainWindow.MyEvicConfigurationButton=myEvic konfigurieren\n(derzeit
inaktiv)

All other language packages please also adjust.

nbraunstein, 2017-02-13:
I can't wait for the configuration tool for Clock Select. I check everyday heh...@clockselect

MicroSur, 2018-04-08:
Full featured SME does not require configuration tool.
https://github.com/MicroSur/sur_myevic