ClockSelect/myevic #279: Mod doesn't shut down after saver

javak810i, 2017-02-14:
Hey, I have a small issue with the latest version of this CFW. My mod never shuts down even after the saver ends and many minutes pass. I can still wake it up by pressing the fire button even after like 5 hours. 

Box Model: Joyetech Evic vtc Vtwo

Coil setup: Tobeco super tank mini coil 0.5 ohm

fOmey, 2017-02-14:
I believe saver simply turns off the screen.. goes into a sleep mode.

javak810i, 2017-02-14:
From the document:
"duration configured with the "Saver" option, after what the box shuts down.
If the "Saver" item is set to "Off", the box will dim after "Main" second, then switch to a black screen for 3 minutes before being shut down."
Shut down =/= sleep
I thought shut down means the same as pressing the fire button five times

fOmey, 2017-02-14:
I may be wrong, I understand your confusion.. although I've played around with the setting and it seems to toggle the screen timeout delay.. definately does not turn the mod off for me either.

fOmey, 2017-02-14:
I stand corrected.. I just turned it off and your right the setting makes no difference ! Sorry for the confusion.

javak810i, 2017-02-18:
Any comment on this issue?