ClockSelect/myevic #284: Thanks & feature request.

3rd1st, 2017-02-16:
Thanks for your excellent work. This has increased the usefulness of my device immensely(eVic VTwo mini).

I have seen a few feature requests here, like flappy bird in the click menu. I'm definitely hoping to see more custom options in the "click" menu. 

My personal request is for "invert screen" available as an option under "click" menu... for when I want to keep a close look at my voltage drop/ battery sag when pulling on vape.

Again, thanks for all your work!

arafey, 2017-02-16:
This feature already exists in the preferences under the "Screen" menu. There's an option called "Invert" that you can toggle on and off. You enter the preferences by holding down the fire and + button simultaneously.

int2001, 2017-05-08:
Extending this Feature-Request:
Please add an option to "flip display 180°" only when vaping.

arafey, 2017-05-08:
This project isn't being maintained anymore. You should take a look at [some of the forks](https://github.com/ClockSelect/myevic/network).

MicroSur, 2018-04-08:
Implemented in SME
https://github.com/MicroSur/sur_myevic