ClockSelect/myevic #288: RX 2/3 cant upload logo not 64x48

HarmoniousInfinity, 2017-02-18:
**Box Model:**

**Coil setup:**

HarmoniousInfinity, 2017-02-19:
Hello. I can't upload logo not 64x48 with original tool even after your firmware on my device. And after try it device go offline and not power on while uploading firmware again. After that again can't upload logo not 64x48. What i can do for resolve this problem?

fOmey, 2017-02-19:
I own a RX2/3 and just tested changing logo's worked fine using "Update Wismec 1.00".

HarmoniousInfinity, 2017-02-19:
dont work for me with other resolution than 64x48 (screenshot http://c2n.me/3HFlwOM)

fOmey, 2017-02-21:
That is because if you want to use any other resolution you must upload a "splash logo". Use NFirmwareEditor and edit the very last resource. Should be a blank screen by default.

HarmoniousInfinity, 2017-02-21:
oh. great. its working. BUT not resolving my problem ) i want to logo, when set at mid position, dont overlap third line partially (screenshot http://c2n.me/3HJr18C)! for that i need to logo have 64x40 or 64x52 resolution. because i didnt find how set splash logo to not screensaver only.

fOmey, 2017-02-21:
Whats the objective ? You want to centre the logo and hide the stats behind it ?

HarmoniousInfinity, 2017-02-21:
Wow, yes ) it's what i wanted. Great now. Thank you. Looks like cetered too ) And with NFirmwareEditor i could upload splash logo with 64x128 resolution and working good too (for 'hello screen').

fOmey, 2017-02-21:
Awesome, glad you got it sorted.

If your satisfied with the fix consider closing the issue.