ClockSelect/myevic #290: Preheat with SS316 TC?

photog0411, 2017-02-23:
This custom firmware you made states that preheat only works with POWER or SMART mode. is this true?

The new EVIC MINI official firmware has a new preheat function that works with ss316 temperature control mode. Can you add it to this custom firmware?

I'd really like to install this custom firmware, but I cannot live without the preheat function for my SS316 fused clapton coils. I'd really like to ditch the official firmware.

Actually if you can add this feature in I'll paypal you 10 dollars as a bribe

photog0411, 2017-02-23:
I see two other features that look like they do something "similar"

PPwr On/Off
and
Boost

Is that like a preheat?

nbraunstein, 2017-02-26:
No, it isn't. One is menu functionality, other reduces oscillations.