ClockSelect/myevic #291: [Bug] Uneven battery drain in RX300

Lizzardis, 2017-02-23:
**Box Model:**
RX300

**Coil setup:**
TFV8 w/ 10 Coil @ 0.13ohm.

I'm just writing to report a possible bug in the latest myevic firmware.

I have been using MyEvic on my RX series mods for a while now, and have never noticed any problems, and infact have always enjoyed this firmware so much better than stock.
So naturally, when I received my RX300, the first thing I did was update it to MyEvic.

After a couple of hours though, I begun to notice something strange.

The battery percentage on the main screen was draining much quicker than it should be, and whilst I know that the battery percentage/icon on the main screen just shows you the lowest volts out of the set of batteries, it was decreasing far too quickly.
Checking the volts in the BVO setting, I noticed that B2 was always at least .2v lower than the other 3 batteries. 
So, noticing this, I swapped out the batteries for different ones that I'd just taken off charge. As usual, BVO reported that they were all at around 4.2v.
Again however, after just a couple of hours of vaping, B2 was hovering .2v lower than the other 3, so whilst B1, B3 & B4 would be at 4v, B2 is at about 3.8.

After testing this problem some more, with different married & unmarried batteries, I always noticed the same pattern emerged. So, I checked my mod. It being brand new, I didn't notice any problems with the connections, so in my defeat, I reinstalled the latest stock RX300 firmware.

Due to the stock firmware showing me the charge levels of all 4 batteries on the front screen, I was able to keep my eye on the batteries over the next couple of hours. Unfortunately, I noticed that all 4 batteries were draining evenly whilst using the stock firmware, which is of course good, but bad in the sense that it appears there was something wrong with MyEvic.

I will note that whilst vaping, I was at roughly 95W with a preheat of 110% for 1 second, so it's not as if I was vaping at incredibly high wattages which would drain my batteries incredibly fast.

I'm just wondering if there could be a fix for this bug? Or even if anybody else has suffered with the same thing?

I can provide any information you need and will help out the best I can. I hate reporting bugs, but if it will better improve MyEvic, then it is totally worth it.

fOmey, 2017-02-23:
I experienced a similar situation on my RX2/3, I was actually in the process of trying to make the battery voltage readings more accurate. 

Unfortunately tho my RX2/3 died yesterday.. very poor design choices on wismecs behalf on this device but I digress..

I noticed the exact same thing with my RX2/3, as the battery drained the voltage readings became less and less accurate, I had tested the battery voltages throughout the battery discharge and even tho they were being reported .2, .3, or even .10 lower than each other they were indeed being discharged evenly.

This is the ReadBatteryVoltage() function for the RX2/3:
`			VbatSample1 = 125 * ( VbatSample1 >> 7 ) / 100;
			VbatSample2 = 139 * ( VbatSample2 >> 4 ) / 624;
			VbatSample3 =   3 * ( VbatSample3 >> 4 ) >> 3;

			BattVolts[0] = VbatSample1;

			if ( VbatSample2 <= VbatSample1 + 20 )
				BattVolts[1] = 0;
			else
				BattVolts[1] = VbatSample2 - BattVolts[0];

			if ( VbatSample3 <= VbatSample2 + 20 )
				BattVolts[2] = 0;
			else
				BattVolts[2] = VbatSample3 - BattVolts[0] - BattVolts[1];
`

Not sure how code savy you are, but there has got to be a better way to calculate battery voltage then the current solution.. Because were subtracting voltages from one another they cant truly be accurate, 

I actually got very accurate readings at one point while testing but it opened up another issue regarding 2 or 3 battery mode at least on the RX2/3.. the problem I discovered was that each sample will return a value regardless how many batteries you have in the device at any given time, I was in the process of sorting through the values to attempt to map out certain battery configurations.. but as I mention my RX2/3 had died and I wont be able to continue till I find a replacement device.

MrScaryZ, 2017-02-23:
I have exactly the same problem with the RX 300 it is draining batteries
unevenly and this has been checked on a OPUS 3100 V 2-2 and it occurs with
both LG HE4 and Samsung 25R's in fact it drained battery #4 to 2.5 volts
several times

lostvoid, 2017-02-27:
I've got the same issue on my 300. I've also noticed that there is an issue with the device in standby mode where it will drain B2 to 0 volts and kill the battery.

USA-RedDragon, 2017-03-07:
Can confirm as well

danielpublic, 2017-03-08:
Not sure yet if I can confirm this on my rx300, but testing [Game to click menu & multiple battery voltage stats #285](https://github.com/ClockSelect/myevic/pull/285) I noticed on batterys V were really really low.

Lizzardis, 2017-03-08:
So it seems that this is a genuine bug which effects a fair few of us RX300 users, and isn't necessarily down to any brand of battery or way of using the device.

From the evidence given above, it's the firmware falsely reporting the battery voltage levels and making it so that 1 of the 4 is draining much quicker. Or at least, the firmware is responsible for the over-draining of 1 out of the 4 batteries.

Am I right in saying that?

If so, is there any way to fix this? As @fOmey suggested, there must be a better way to calculate the voltages, however I'm at a complete loss due to my lack of coding skills. I just hope that at least with myself bringing attention to it, and others reporting that they too have problems, that it can somehow be fixed.

fOmey, 2017-03-08:
Can you guys please check your battery voltage with a multimeter, voltage checker, charger or what ever and figure out if the battery voltage is just being reported with incorrect voltages or if the battery's are indeed being discharged unbalanced.

My RX2/3 was reporting unbalanced voltages, although my battery's were discharging at the same rate.. so the firmware was just reporting the incorrect voltages... 

Unfortunately I do not own a RX300, also my RX2/3 died recently.. I wont be able to continue development until I get my hands on a new device.

Hopefully @ClockSelect can chime in, I'm sure he would have an idea of the best way to resolve this.

USA-RedDragon, 2017-03-08:
After I noticed battery drain on one battery I immediately switched back to
stock firmware where voltage persisted to be uneven.

shurizzle, 2017-03-09:
Same here with a RX2/3

danielpublic, 2017-03-10:
The rx300 threw an 'check battery' at me, put the batteries in my bt-c3100. It displayed: 4.08v, 4.10v, 3.13v, 4.09v. Can post an pic later, since my phone doesn't seem to cooperate. Charging batteries atm.

MrScaryZ, 2017-03-10:
Has their been any progress on this? I had to revert back again to the
stock firmware. Uneven drain is one thing but draining a battery to below
 Liion safe specs is another.. now it has happened with HE4's HG2's VTC5a's
and 25r's

Lizzardis, 2017-03-10:
[image]
[image]

Lizzardis, 2017-03-10:
*Apologies for closing & reopening the thread, I was trying to sort out some pictures uploading. Heh. Eventually had to resort to uploading to Imgur due to them not uploading through Github.*

Okay so, I've just reinstalled myevic onto my RX300 to test this out so you guys have more data on your hands. You'll have to give me a couple of hours, but I'll try and be a thorough as I can.

I've literally just this second reinstalled myevic and I'm about to pull 4 Samsung 30Qs off charge to put in the device, so I KNOW they are all at 4.20v.

After putting the batteries in, I can see they're reading a little bit off kilter, which is fine.

http://imgur.com/kW2tvyI

I've effectively "zeroed" the batteries to reading at 4.20v, so they're all as equal as possible.

http://imgur.com/UwsjhCw

And I will be vaping at 95w with a 110% preheat for 0.8 seconds.

http://imgur.com/06rIZd6

I will report back in a couple of hours with the results!

ayhancham, 2017-03-10:
First things, I am soryy for my english. Yestarday I make like same test with RX2/3. I think that device only use  one battery, when screen open,for board or for control board temp. etc. . maybe for this just one battery go down quicky.

Lizzardis, 2017-03-10:
I'm back!

I was going to leave it a little longer, perhaps 6 hours, but after checking the BVO values, it's quite clear the pattern that's emerging.

From the homescreen you can see the battery drain.

http://imgur.com/zYHZ8ed

But the BVO settings really come through here. As you can see, B2 is significantly more drained than the other 3 batteries.

http://imgur.com/WEF38Zi

And when I put that battery into my charging unit to check the voltage, it comes out even more drained than myevic was reporting!

http://imgur.com/wEYiPXl

It's strange that it's just that one battery though. Is perhaps that battery responsible for preheat, or something?

Either way, I'll keep myevic on my device for now and come back in a couple of hours with more results.

fOmey, 2017-03-10:
@Lizzardis, Can you post your actual battery voltage also, not what myevic is reporting... It's obvious the voltage myevic is reporting is not accurate as mentioned above..

Please check that your battery's are not actually discharging unbalanced.

Lizzardis, 2017-03-10:
Check the 3rd link in my last comment,, that's the voltage for B2 that my charging unit is showing. The voltages for myevic is the 2nd picture in my above comment.

http://imgur.com/wEYiPXl

Lizzardis, 2017-03-11:
For pure clarification:

B1

http://imgur.com/AK7TB2L

B2

http://imgur.com/wArKV3p

B3

http://imgur.com/1Z8KeWU

B4

http://imgur.com/I0QqDUI

fOmey, 2017-03-11:
Interesting, so they are indeed discharging unevenly..

Lizzardis, 2017-03-11:
It's interesting that it seems to always be B2 as well, only when there is 4 batteries in the mod. If it's a 3 battery mod, such as the RX300, all batteries discharge evenly.

escoltajuverf, 2017-03-27:
this bug only kicks in after you flash to different firmware and or reset dataflash using nfe tool, happened to me with 2 rx 300 mods, there must be some calibration done in the factory, i am hoping someone can upload untouched stock dataflash file from rx300, gonna try uploading the stock dataflash to the device to check if mod reads battery voltage correctly, this incorrect voltage reading affects both charging and discharging mainly bat number2

USA-RedDragon, 2017-04-04:
https://github.com/USA-RedDragon/myevic/commit/a5aa1d4b126892202035672d5ef496d459fdfef9