ClockSelect/myevic #292: Evic Primo Support

fOmey, 2017-02-24:
As the title suggests, evic primo support.

**Box Model:**
Nothing at the moment :(

**Coil setup:**
0.16 fused.

fOmey, 2017-02-24:
Disregard, not worth the time considering it had 510 issues..

andreivoinea93, 2017-02-24:
Only the first couple of hundred units had that problem. They fixed it.

Here is mine

https://goo.gl/photos/b1n5ixn2nD7N1mHp7

dywersant, 2017-02-25:
All Jt, Wismec etc products have 510 issues :)

andreivoinea93, 2017-02-25:
Did you even bother to watch the video? :)

dywersant, 2017-02-25:
I have watched the video, yeap.
Here is mine :D [xxx](https://i.snag.gy/XgJ2YC.jpg)

fOmey, 2017-02-25:
The 510 in his video is spring loaded, meaning Joyetech have finally learnt ! HALLELUJAH!!!

I'm about to do the same to my RX2/3 once my varitube 510 comes in.

andreivoinea93, 2017-02-25:
> Although I most likely will purchase a primo.

I have it for about a week now and can't find any cons for it. Seriously. We'll see how it does with time though.