ClockSelect/myevic #294: Please support eVic Primo

MicheleLynnBass, 2017-02-26:
@ClockSelect Please, please, please, make compatible with eVic Primo.  Love the mod, hate the original firmware on it.

andreivoinea93, 2017-02-26:
I put ArcticFox on mine until @ClockSelect does the support for Primo. Will move to myEvic once the support is here because it's opensource. But meanwhile, you can try AF.

MicheleLynnBass, 2017-02-26:
I'm not familiar with ArcticFox...do you have a link?

andreivoinea93, 2017-02-26:
http://nfeteam.org/

MicheleLynnBass, 2017-02-26:
Thanks, I'll check it out.

MicheleLynnBass, 2017-02-26:
Cannot find version of ArcticFox that will upload to my Primo...keep getting error.

fOmey, 2017-02-26:
Duplicate of issue #293, articfox primo support is avalible on the nightlies. Let's stay on topic tho please, this is not a support section for another firmware.

MicheleLynnBass, 2017-02-26:
Thanks and sorry.

GravityRoller, 2017-04-17:
With release of eVic Primo v2.0 hoping for Porting support to run this firmware. Any idea if/when this may happen?

BigBadBlackWolf, 2017-04-18:
ClockSelect is probably no longer available for the further development /
improvement - why ever. MiroSur has apparently accepted the case. For more
information see:
https://github.com/MicroSur/myevic/find/master?q=
Last current version of myevic is 3.30 from 09-04-2017. With Flappy Bird
and Tetris.