ClockSelect/myevic #296: Please add USB REVERSE CHARGING :)))

HAQER, 2017-02-28:
🔌🔌🔌🔌
Hi. 
I'm very satisfied of your hard work with this software, but i have a suggestion.
You see new function from **Evic VTC Primo**? Yeah! It's "reverse charging" with special adapter (available here: http://www.wismec.com/accessories/avatar-reverse-charging-adapter/ ). [also on aliexpress 😄 ]
 
Can you add this option to this software? Then we can use box as powerbank with this adapter.
I think it's possible and very usable to every user who buy this adapter.

docdeath, 2017-02-28:
its not worth it if you need power bank buy one   its just selling point and not stable  tested my self with there own   firmware  and will drain mod battery's very fast and that's on rx 300 and  this has 4 cells  so on any mod with less battery,s forget it

ClockSelect, 2017-08-30:
WHO will drain the e-cig battery to power something else?
I'd prefer my phone to power my e-cig, my pad to power my e-cig, my car battery to power my e-cig, than any of the reverse.
Why would you like to charge your phone with your e-cig? To be able to use it to know where some café nearby will let you charge your e-cig?
And NO, you can't start you car, even with a RX300.