ClockSelect/myevic #297: Kicked out of TC at 68% battery

yassert, 2017-03-01:
Hi,

First of all, thank you for your work. I have been enjoying your firmware quite a lot.
As the title states, I have an issue with temp control. I get kicked out of TC mode into power mode when the battery level reaches 68%.

I have tried the following, but nothing has worked:
1. Changing the battery.
2. Turning the mod off with five clicks and then turning it on again.
3. Taking the battery out and putting it back in.
4. Updating the firmware to the latest binary available here.

I was wondering if you had an idea what was happening? Am I the only one experiencing this issue? 
Thanks.

**Box Model:** VTC Evic Mini  

**Coil setup:** SS, single coil, spaced, 0.8 ohms

**Vaping mode**: Temp control

lojpek, 2017-03-02:
I think that your accu have some max Amp and when you vape on TC sometimes that Amp limit is exceeded. If your box take too much Amp compare to your battery then some algo change to power mode. And when your coil have some issue then box will change that mode too.

yassert, 2017-03-02:
I have tried different batteries (Samsung ICRs rated for 28A) and a different type of battery (a VTC5). 

I reduced the temperature setting but it did not help. I am using TC in default settings so there is no boost etc.

lojpek, 2017-03-02:
Try to change setup. I have sometimes that issue and when I  change setup it's OK.

yassert, 2017-03-02:
OK, I will try. It is strange though, because if I put in a fully charged battery, TC works normally.

dywersant, 2017-03-03:
Disable tc coil check in menu. Details can be found in the manual.

Raybo58, 2017-05-28:
I'm having a similar issue. When I try to use the Ni TC mode, it operates for about 1.5 seconds and then kicks over to power mode. It doesn't seem to be related to the battery level, as in yassert's case.

If I turn off tc coil check, then i get kicked into power mode instantly. In fact, with coil check turned off, it messes with even selecting TC mode. I do the three clicks, use the plus button to highlight the TC Ni mode, but when I press fire to select it, it instantly goes to power mode. The only way I can activate TC mode with coil check off is to highlight it and then wait for ten seconds for it to become active. But then it just acts as it did before, reverting to power mode after about 1.5 seconds.

**Box Model:** Reuleaux RX200S

**Coil setup:** Wismec Rx Triple Coil 0.15Ohm (NiChrome)

**Vaping mode:** Ni Temp Control