ClockSelect/myevic #300: Request: Add support for Wismec Predator 228

Justin-L-Franks, 2017-03-10:
The new Wismec Predator 228 looks to be based on the same chipset that all of the other mods that myevic will run on (same screen, same factory UI as the RX300, etc.). It's a double 18650 mod that competes directly with the Smok Alien 220W (hence the name Predator....clever!).

Will you be adding support for this mod? My original Cuboid is getting old, and is scratched and chipped to hell; it still functions perfectly but I'll be getting the Predator if it will run myevic, and delegating my Cuboid to a backup mod. Otherwise I'll be getting either a Wismec RX2/3 or Presa TC100W.

Thanks!

fOmey, 2017-03-10:
I second this, stay away from the RX2/3.. horrible device.

I intend on getting a predator myself, would love to see myevic on there.

Justin-L-Franks, 2017-03-10:
@fOmey I'm assuming you're referring to the 510 connection on the RX2/3? My original Cuboid has the same one, and it's a year old with heavy usage and I change to a different atomizer at least a dozen times a day, and I haven't had a single issue. But I was careful from the get-go, and never worried about a fraction of a millimeter of space preventing the atomizer from sitting flush and not wrenching it down tight (which seems to be what messes up the 510 connection).

Other than that, the RX2/3 generally has excellent reviews. Some people complain that it's a few millimeters taller than the RX200/RX200S, but that is pretty nitpicky. Unless there is something that I'm missing?

fOmey, 2017-03-10:
The 510 is a big part of the issue.. although not my only issue.

They use a extremely thin gauge wire attached to one of the battery contacts which is obviously spring loaded, by removing your battery's you slowly weaken the joint and eventually snaps.. renders 2 battery mode useless and forces you either to use 3 battery mode or tear the mod apart and resolder..

Very flawed design in my opinion, also if your battery cover comes off while your device is connected to a USB it pretty much bricks the device.. this may seem like a very rare occurrence, although while I was coding and testing some myevic source. I removed my battery door and rendered my device useless after that, ever since that moment my RX reports "no atomizer", I even went as far as to replace the 510 with a varitube 510 being under the impression the 510 must of died... later discovered the chip had bugged out due to the battery door being removed. This has been reported by a few other users on reddit.

It's a true shame, really is.. the concept behind the device is excellent and I thoroughly enjoyed using it, although it could of been planned out a little better to say the least.

dywersant, 2017-03-10:
I secon that. Few days ago my friends rx23 stopped working. It turned out that the bottom battery contact wire broke. The other thing is that wire was laughably thin, around 24Ga i presume. I got it replaced and soldered. 
If you want something cheap and well made i suggest Sigelei Fuchai, firmware can be modded as well.

javak810i, 2017-04-08:
Any news about the support for Wismec Predator? I heard only good things about it so far and I will be buying it. I would welcome the myevic firmware because it's superb.