ClockSelect/myevic #304: [Question] Porting

Dil3mm4, 2017-03-14:
Hello there, are there some guide/help lines to port this firmware to another device? 
Is there a proper documentation for devs about this firmware? 
Thank you for your time.

fOmey, 2017-03-17:
I'm sure you have already but take a look at the commits for new devices, it will give you an idea what is involved. https://github.com/ClockSelect/myevic/commit/6bdf4b343d0a2cd678a9fc7f5ef84beba089af3b

ClockSelect, 2017-08-30:
I can't, really. Any new device may have some surprises, and I burned some boxes learning it. I won't make a do-it-yourself doc for that, since I feel somewhat responsible for what I do and for for what people do with it. I won't provide a documentation that may not fit the next device about wich I know nothing.

closed 2017-08-30 by ClockSelect

Dil3mm4, 2017-08-30:
That's clever.
And I really appreciate you for that. 
Can you take a look on my latest issue?
Would be nice to hear words from you. :)

ClockSelect, 2017-08-30:
Are you talking about #337 ?
(please give a link when you talk about another issue)

Dil3mm4, 2017-08-30:
Yes! :)