ClockSelect/myevic #306: new version

tiryaki92, 2017-03-17:
When will the new version come and what changes will happen

nbraunstein, 2017-03-20:
I was wondering the same thing...

tiryaki92, 2017-03-22:
fomey or clockselect please answer

tiryaki92, 2017-03-22:
@ClockSelect  @FOmey

nbraunstein, 2017-03-22:
Just use Arctic Fox, it works great.   
    
  
 Sent from my iPhone, please excuse any syntax errors.         

>   
>   
>   
>   
>
> @ClockSelect (https://github.com/ClockSelect)   @FOmey (https://github.com/FOmey)

 ]]>

enricorov, 2017-03-22:
I'd like to use this firmware, I like this one better. However, asking for ETAs is rude. @ClockSelect will reply when he can.

nbraunstein, 2017-03-22:
I don't feel it's rude to ask for an ETA, but that's just me.    I enjoy both versions of the firmware, I'll use one and switch back once ClockSelect is updated. No big.   
  

    
  
 Sent from my iPhone, please excuse any syntax errors.         

>   
>   
>   
>   
>
> I'd like to use this firmware, I like this one better. However, asking for ETAs is rude.  @ClockSelect (https://github.com/ClockSelect)  will reply when he can.

 ]]>

messywrk, 2017-03-30:
Dont want to assume the worst, but given the recent history its possible that @ClockSelect is not actively developing anymore.. :| Is anyone willing to share some knowledge, maybe in the Wiki or so?

dabugen, 2017-04-25:
I´ve heard that ClockSelect is dead, so the project is over....

BigBadBlackWolf, 2017-04-26:
The project myevic is no longer developed by ClockSelect for a long time -
why always. For this, MicroSur has accepted the matter and a new update
V3.30 (14-04-2017), which contains the game Flappy Bird also the game
Tetris. https://github.com/MicroSur/myevic/find/
<https://github.com/MicroSur/myevic/find/master?q=>master?q=
Beside myevic there is also the project openevic of Abuzer Rafey, which
still seems to be confused with many bugs, since a lot of users with their
e-cigs problems with this firmware and the download page https
://github.com/arafey/openevic <https://github.com/arafey/openevic> is now
no longer available.