ClockSelect/myevic #307: Bug in Preheat menu (selecting % or W)

Justin-L-Franks, 2017-03-18:
On my Cuboid (I have no other compatible mods to test), I found an issue with the Preheat menu. If you try to switch the Unit (percentage or wattage) using the + button, the unit changes, but you get "stuck" and cannot move the cursor to any other option. If you click the + button again, the unit switches back to what it was before, and you can move the cursor to the other options.

So right now, the only way to change between percentage based and wattage based preheat is to change the unit, then press + and - simultaneously to move up to the Vaping menu. Then you need to go back into the Preheat menu to change the other settings.

**Box Model:** Cuboid v1.01

360up, 2017-03-31:
confirmed bug on vtc-mini

ClockSelect, 2017-08-30:
No news, good news.

closed 2017-08-30 by ClockSelect