ClockSelect/myevic #310: Battery discharge/charge % incorrect in eVic VTwo (inbuilt battery)

adid45, 2017-03-30:
I have set the battery to "generic" in the expert section but sometimes the battery % starts discharging rapidly (like 4% per drag) until it reaches zero. After connecting it to the charger the battery % increases quickly to level which it was before starting to discharge at this high rate. 

I am using an Evic Vtwo @35W on 0.2ohm coils.

ClockSelect, 2017-07-25:
The VTwo battery profile hasn't been taken into account; I'll fix that.

But it won't solve your problem if your battery is dead.
With the generic battery profile, the VTwo should seem to drain slower, not faster.

adid45, 2017-07-25:
The battery is alright, I am still using it on stock firmware.

I will test again this weekend.

ClockSelect, 2017-08-30:
No news, good news.

closed 2017-08-30 by ClockSelect