ClockSelect/myevic #315: Add profiles to curve settings

lojpek, 2017-04-10:
Hello.
I think good move will be add curve profiles do curve in menu (select using curve and settings for curve1, curve2, curve3, curve4, curve5).
I think that profile manage is not good without erase profile setup but I think that couple curve settings is good for searching some good curve without writing it on paper and editing on box and searching good enought or one for warm vape and some for cold vape or other settings. 

Regards.

Wismec Presa 75W, Avocado24 RDTA, coil from SS mesh200 and cotton wick.

BigBadBlackWolf, 2017-04-18:
ClockSelect is probably no longer available for the further development /
improvement - why ever. MiroSur has apparently accepted the case. For more
information see:
https://github.com/MicroSur/myevic/find/master?q=
Last current version of myevic is 3.30 from 09-04-2017. With Flappy Bird
and Tetris.

fOmey, 2017-04-18:
@BigBadBlackWolf , Please stop spamming every issue with that message.. I'll have to unsubscribe from this repo if you continue.

ClockSelect, 2017-07-25:
Profile deletion is now available, and curve is saved in the profile.
You can use profile switching to switch between duplicate profiles with different curves.

closed 2017-08-30 by ClockSelect