ClockSelect/myevic #317: no more versions?

nbraunstein, 2017-04-17:
is this project over?

**Box Model:**

**Coil setup:**

easyyu, 2017-04-17:
It look like yes..

BigBadBlackWolf, 2017-04-17:
ClockSelect is probably no longer available for the further development /
improvement - why ever. MiroSur has apparently accepted the case. For more
information see:  https://github.com/MicroSur/myevic/find/master?q=  Latest
current version of myevic is 3.30 from 09-04-2017. Without game Flappy Bird
and Tetris.

BigBadBlackWolf, 2017-04-17:
sorry, with the games Flappy Bird and Tetris.

Am 17.04.2017 13:28 schrieb "Unbekannt Niemand" <unbekannt.niemand@gmail.com
>:

> ClockSelect is probably no longer available for the further development /
> improvement - why ever. MiroSur has apparently accepted the case. For more
> information see: [...]

Tolga83, 2017-04-17:
@BigBadBlackWolf thank you very much for the update, we all were waiting for clockselect to keep the project carry on. But nobody was told us that he quit from it. So I jumped into MicroSur's page thank you again.

dabugen, 2017-04-25:
I´ve heard that ClockSelect is dead, so the project will be over, unfortunately.

enricorov, 2017-04-25:
@dabugen he's.. dead?

nbraunstein, 2017-04-25:
Project is dead or he is dead?
    
  
 Sent from my iPhone, please excuse any syntax errors.

ClockSelect, 2017-07-25:
Let's say it's in slow motion.