ClockSelect/myevic #318: NEW VERSION AVAILABLE

tiryaki92, 2017-04-18:
Clockselect will no longer continue. We will continue from here. Support identifying new features and issues.

https://github.com/arafey/openevic

dabugen, 2017-04-25:
Nice, but the link is not working, 404 error....

vanita5, 2017-04-26:
Looks like it's dead already. I still have their latest binary (0.15) if anyone still wants to use it.

BigBadBlackWolf, 2017-04-26:
Project openevic has to many bugs. User have a lot of problems with their
e-cigs. MicroSur has apparently accepted the case from ClockSelect. For
more information see:  https://github.com/MicroSur/myevic/find/master?q=
Latest current version of myevic is 3.30 from 09-04-2017. With the games
Flappy Bird and Tetris.

ward0, 2017-04-28:
Why should i upgrade from the standard firmware? Ok a clock would be nice to have.. But having something in my mouth and bugs in the firmware... You know...

BigBadBlackWolf, 2017-04-28:
You have more settings in the custom firmware compared to the stock
firmware. Read the corresponding Readme.md in the indicated link to myevic
from MicroSur. As a normal user you do not need the custom firmware
(absolutely) and can use the standard firmware.

StylishSeahorse, 2017-05-28:
Vanita5 if you still have the latest binary I would be very interested

BigBadBlackWolf, 2017-05-28:
The last myevic version is from 26/04/2017. Download from here:
https://github.com/MicroSur/myevic/blob/master/bin/myevic.bin

StylishSeahorse, 2017-05-28:
Thanks I have that already. However am looking for the latest version of openevic, which I believe can't be downloaded anymore.

BigBadBlackWolf, 2017-05-29:
openevic and myevic (from MicroSur) based on myevic (from ClockSelect).
However, there were probably with some users of openevic many problems with
their e-cigs, which somehow could not be remedied by the programmers, so
they have the project discontinued. In addition to the latest version of
myevic (from MicroSur), many of them have also moved to ArcticFox (from
NFE-Team). I can only advise you to also leave the fingers of the last
version 0.15 of openevic.