ClockSelect/myevic #320: How to put and create custom logos into the firmware?

the27club, 2017-04-24:
I am just wanting to create a logo to add on to my RX200s, but still want to keep the firmware.
 
Is this possible? 
please excuse my ignorance if there is..

BigBadBlackWolf, 2017-04-24:
Yes, it is.

the27club, 2017-04-25:
Yeah. How do you do it though?

BigBadBlackWolf, 2017-04-25:
Use a graphics program with which you can create 1-bit images.
Then create your picture in the appropriate resolution (64 x 40 px, 96 dot)
and save it as .bmp file.
Take the FwUpdater from Wismec and select LOGO button, now select your
created logo and it will be added.

the27club, 2017-04-25:
so I did this and it doesn't cover the clock. I see the logo but the bottom part of the clock still shows. Logo can't be over 64x48

BigBadBlackWolf, 2017-04-25:
This is true, the logo can not go over the entire screen resolution. The
Wismec RX 200S has max. 64 x 48 px for the logo. In the menu under Screen
at Logo, select Where on "Top"

fOmey, 2017-04-25:
Post a picture please.

MicroSur, 2017-04-29:
Try my fork, I step on this shit and fix week ago.

djmajo, 2017-05-14:
www.vaping-pixels.com

I buy mine here.... 1 Buck each logo. And they are really good and handmade!
Love them!