ClockSelect/myevic #322: Very important

sigaraci, 2017-04-25:
Everyone, hello. Thank you to everyone who worked in this excellent project. This software is a revolution in electronic cigarette software. But I see that the software is not updated due to some reasons. This revolution will continue if people with previous contributions to this software can continue this software. Also please let us know why you are not updating the ones connected with ClockSelect. Thank you @fOmey @USA-RedDragon @ClockSelect

BigBadBlackWolf, 2017-04-26:
The project myevic is no longer developed by ClockSelect for a long time -
why always. For this, MicroSur has accepted the matter and a new update
V3.30 (14-04-2017), which contains the game Flappy Bird also the game
Tetris. https://github.com/MicroSur/myevic/find/master?q=
Beside myevic there is also the project openevic of Abuzer Rafey, which
still seems to be confused with many bugs, since a lot of users with their
e-cigs problems with this firmware and the download page
https://github.com/arafey/openevic. Is now no longer available.

Lizzardis, 2017-04-30:
Does the EVic firmware being developed by MicroSur fix bugs which have been reported recently? Such as the uneven battery drain on the RX300? #291

fOmey, 2017-04-30:
I no longer own any regulated devices with a compatible chip. I may consider buying a compatible device in the near future.

USA-RedDragon, 2017-04-30:
I dropped the change on the issue page https://github.com/USA-RedDragon/myevic/commit/a5aa1d4b126892202035672d5ef496d459fdfef9

Lizzardis, 2017-05-01:
@USA-RedDragon Wait, so does that mean the issue might be fixed?

fOmey, 2017-05-01:
Yes, try his fork lizzardis.

USA-RedDragon, 2017-05-01:
I've been running my fork of myevic for months now. No weird drain but the
voltage reading is off. Putting the batteries in a charger shows they're
fine

Lizzardis, 2017-05-01:
Superb, so as long as I correct the voltage reading via BVO, I wont have the single battery drain of B2 or B3? As in, one battery shouldn't drain extremely more than any other?

fOmey, 2017-05-01:
Let me clarify, there are two separate issues with myevic currently.

1. The wrong battery voltage is being reported by the firmware, it's not 100% accurate.. if your using a firmware with my multi-cell voltage display commit it's much more obvious. However its a false reading, the battery's are not actually being discharged unevenly !

2. The RX300 however has been reported that it actually does discharge one of it's cell's unevenly.. which is what the commit USA-RedDragon referenced in his previous reply fixes.

To summarize, there is no fix for the incorrect voltage readings yet.. however your battery's are not going to be discharging unevenly anyway so its not a major issue. If you own a RX300, it's highly recommend you use a firmware with the commit USA-RedDragon referenced as this fixes.

AlexFundorin, 2017-06-19:
There's still ArcticFox available. https://nfeteam.org/

USA-RedDragon, 2017-06-19:
@AlexFundorin The issue some of us have with ArcticFox is that it's not open source. Developers like me enjoy modifying the source ourselves

USA-RedDragon, 2017-06-19:
And don't you know ArcticFox has all the spyware in there source? They track all out vaping and send it to the government. /sarcasm

AlexFundorin, 2017-06-19:
Of cause, I'm aware of it's nature. We, the russians, always spy on the rest of the world. You can't fight it, so, surrender immediately. All your base are belong to us.

As a user, I'm fine with AF, not being open source, while it gives me option that I need from my vaping device and lots more. Since myevic is abandoned, AF is the best firmware around, ATM.

USA-RedDragon, 2017-06-19:
I mean there's nothing wrong with myevic. I'm using a build of it with the quad battery fix that I've had for weeks and I've been running myevic since a month after the rx300 come out. I don't have the prowess of ClockSelect to move it to new devices sadly, so that will have to be someone else. And in that scenario where I get a device that myevic doesn't run on, I'll have to switch to AF

closed 2017-08-30 by ClockSelect