ClockSelect/myevic #325: I need the fattest vape clouds

jamesalbert, 2017-05-14:
Hi, quick question. I love this firmware, but I'm not too keen on the vape jargon. I noticed there's an "expert" tab in the menu to change certain settings and I was wondering if there's a settings configuration that allow for thicker whiter clouds? Not sure if this makes sense, but I keep seeing videos on youtube of people vaping really thick white clouds and I'm just wonder if that's software or hardware issue?

If this is irrelevant to the firmware let me know and I'll close the issue. Thanks in advance!

**Box Model:** joyetech evic2 mini

**Coil setup:** cleito .4

MrScaryZ, 2017-05-14:
That is the dumbest thing I have ever heard... you think clouds are based
on firmware or that a setting turns on big cloudz

MrScaryZ, 2017-05-14:
> I was just asking what settings I might be able to set that could result
> in larger and or more vape. Thanks for the thoughtful, insightful response
> Randy.

James, you may be upset and maybe I did not reply to it in a Vaping 101 way
however, when someone is upgrading firmware of a device the assumption is
they
understand what and how to achieve larger clouds..
You could do a preheat or one of the curves to set up so the initial hit on
the coil is larger.. Example on my RDA I will have the preheat curve around
100w starting and then down to 80 for a sec then to 70.. just an example
hope that helps

jamesalbert, 2017-05-14:
I came to this project's page looking for information and you responded rudely. I don't think it has anything to do with vaping 101.

> when someone is upgrading firmware of a device the assumption is they understand what and how to achieve larger clouds

So any idiot off the street who spends $30 on a vape should know everything about vaping? Where's the logic in that?

> You could do a preheat or one of the curves to set up so the initial hit on the coil is larger.. Example on my RDA I will have the preheat curve around 100w starting and then down to 80 for a sec then to 70.. just an example hope that helps

Thanks man, this does help. This is the kind of info I asked for initially. Thanks for providing it.

MrScaryZ, 2017-05-14:
you asked for a setting to blow clouds you are an idiot

jamesalbert, 2017-05-14:
Says the guy with an empty git history who jumps onto github to make rude comments instead of contributions. And now everyone on github knows that Randy Wheeler is a jerk.

enricorov, 2017-05-14:
@jamesalbert you need to understand the limits of a mod. The thickness of the clouds depends, in order of relevance:

- The atomizer you're using, don't expect to have huge amounts of vapor on a 14mm, MTL tank, you need drippers, aka RDAs, for that. 
- The raw power output of the mod you're using, it needs to be capable of delivering the power huge clouds require. For thickness, generally you want to stay at least in the ~100W range, and you need at least two batteries for that. 
- The firmware of the mod, to tweak the power delivery even further 
- Inhaling skills, to breathe in and out the amount of vapor you intend to produce. 

As you can see, you need things other than FW to achieve what you're after. To be perfectly honest with you, go for quality instead of quantity. Big ass clouds do nothing but bother people.

jamesalbert, 2017-05-14:
@enricorov The goal is actually to vape on as many people as possible

No, but seriously, thank you for actually useful information. I thought it was mostly hardware that influences it, I was just wondering what I can do with the settings to increase it. For example, the "expert" tab in the menu, I have no idea what those things do and was wondering if anyone had a personal setup that works for them. But thank you for all the info, much appreciated

enricorov, 2017-05-14:
The guide in myevic homepage has detailed information on the subject. That's a 'Dragons ahead' menu, don't tinker with that. 

The best thing you can do is raise the power output, and play with the power curve, and see what works for you. Ultimately, you'll have to upgrade to see real clouds. I use my VTwo Mini for flavour, and that's where the TC algorithm of this FW shines. 

For raw power, though, a single battery mod just doesn't cut it.

jamesalbert, 2017-05-14:
I didn't think so, thanks anyways 👍

closed 2017-05-14 by jamesalbert

fOmey, 2017-05-14:
Your coil build determines your clouds.. I'm staring at my little stealthy single battery mech mod that I rebuilt yesterday because the clouds were to large and they were embarrassing me at work.. my point being the device you use is irrelevant..

However if your using a 80w device, you want to build lightweight coils.. I wouldn't go over 26g kanthal, that way you have a quick ramp up and "fat clouds".

Throw a sub ohm build, My go to is a 28g fused clapton: 0.20ohm = 4.0v = 20.00amp = 80.00watts.. 5 wraps around a 3mm ID.. perfect for a single battery mod..

jamesalbert, 2017-05-15:
Haha awesome man thank you, this is the info I was hoping to get. I'll stay out of the expert settings, but it looks like I might have to buy a atti that I can rebuild. I'll check all this out!