ClockSelect/myevic #328: Primo support

MicroSur, 2017-07-03:
I`m newbie in C programming but like open source myevic project.
I try to support the primo mod (1 and 2), but I need tests and fixes from professionals.

Take a look at primo branch of https://github.com/MicroSur/sur_myevic

fOmey, 2017-07-03:
Have you actually tested this on a primo ? 

I'm finding your commits a little hard to follow, try leaving detailed comments for changes. Looking at someone else's code and trying to understand reasoning/logic tends to be very difficult.

I own a primo mini and was considering to start working on support, so it's interesting timing that you posted this.. although I must admit my motivation has been a little slack..

MicroSur, 2017-07-04:
I have a Primo v1 and it works. 
I'm looking for a similar code in the firmware for already supported mods.
Marked with a question mark - I'm not sure ...
It's a hobby for me.

fOmey, 2017-07-04:
Ah right, I also have some questions.. I was hoping you would be able to answer. 

I am trying to figure out what PA2, PA3 etc. is used for..

MicroSur, 2017-07-04:
pins to the batteries...

fOmey, 2017-07-04:
Sure, although how do we identify what pins are used for what.. I'm worried
if I declare wrong pins I'll burn out chip you know..

MicroSur, 2017-07-04:
Primo mini InitGPIO code from firmware does not differ from the VTwo mini, except of this
```
	if ( ISRX300 || ISPRIMO1 || ISPRIMO2 || ISPRIMOmini )
	{
		SYS->GPD_MFPL &= ~(SYS_GPD_MFPL_PD0MFP_Msk|SYS_GPD_MFPL_PD1MFP_Msk);
		SYS->GPD_MFPL |= SYS_GPD_MFPL_PD0MFP_GPIO|SYS_GPD_MFPL_PD1MFP_GPIO;
	}
```

fOmey, 2017-07-04:
Right, how did you figure that tho ? I'm still trying to wrap my head around how you figured out battery pins for example.. process of elimination ? Copy and paste current code RX code ? Can you explain how you did it ?

I'm trying to understand how to expand device compatibility, I'll be honest I'm a little lost.. I don't understand how ClockSelect figured out how to declare these pins as output etc. I dont want to have to rely on existing code to get the job done you know ?

MicroSur, 2017-07-05:
Reverse stock firmware of course. With IDA Pro and understanding ARM assembler code. Do You know how to do this?

fOmey, 2017-07-05:
I have no experience with ARM, however I have some experience with PowerPC
from my PS3 modding days.. I would like to pick your brain and have a
proper conversation about this..

If you are willing to give me some guidance would be appreciated.. do you
use discord by any chance ?

MicroSur, 2017-07-05:
Chat with google translate... no thank you ) My native language is russian.

I've been trying to learn ARM for more than a year and a half...
Install IDA (I have v.6.6...), try to load stock firmware, then may be I can help...
https://i.imgur.com/iLhlGJB.jpg

fOmey, 2017-07-05:
Fair enough.. This is what has been troubling me:
[image]

MicroSur, 2017-07-06:
I just press Ok without any changes here.

These are my working files with function names and some comments, it can be useful in learning:
[ida_idb](https://www.dropbox.com/s/tmvbvknvwz9h5ai/ida_idb.zip?dl=1)

fOmey, 2017-07-06:
This is great, thank you very much.

MicroSur, 2017-07-06:
You must decrypt joyetech firmware of course. Use my FWUpdater or NFE to do this first.
[My software here](https://www.dropbox.com/s/83zd19gu05pl3r6/EvicVTCFont.rar?dl=1)

fOmey, 2017-07-06:
I had a feeling it was obfusicated, Now everything has become clear.. thank you.

Haha you must pardon my ignorance, I only started attempting to reverse firmware when I replied to this thread.. so I am learning as I go along. 

Thank you for your help.

closed 2017-07-26 by MicroSur