ClockSelect/myevic #329: Disabling clock on main screen RX2/3?

KranV2, 2017-07-07:
Sorry for posting this here if it's not really an "issue", so I apologize.
But I've gone through almost every single option on my rx2/3 menu and I CANNOT find how to get rid of the large clock on the main screen, that only goes away once you hold the fire button. (Then comes back when I let go), I kinda wanna see my values regardless of button press so any insight on where this is to disable it? Potentially a bug but I'm doubtful, many thanks.

MicroSur, 2017-07-07:
Menu / Interface / Clicks
Choose clock call on fire clicks count, exit menu and then click fire button this times.

KranV2, 2017-07-17:
Many thanks, and apologies for the late reply.

ClockSelect, 2017-07-25:
By default, a double fire click will switch clock display on and off.

closed 2017-07-25 by ClockSelect