ClockSelect/myevic #331: Wismec Predator / Sinous P80 Support

messywrk, 2017-07-25:
Hi Clockselect,

first and foremost, AWESOME you're back! :)

Now, the obvious - would you please add compatibility for the new mods? Can we help you with supplying info/testing?

MicroSur, 2017-07-25:
Great news! and primo mini support )

ClockSelect, 2017-07-26:
@MicroSur Dobry dien :) Grats for all you've done last monthes. I had some success with your Tetris implementation. My talk is something like "See what Russians do when they're given some serious job". ^^

@messywrk I don't know what are the next mods nor when I'll implement them. I'll try to borrow a Primo and with @MicroSur's work it should be easy. Predator seems like a clone of those. I'll see.

MicroSur, 2017-07-26:
Powerbank may be? )

MicroSur, 2017-08-09:
myfork with Predator support, no response from predator users yet...
https://github.com/MicroSur/sur_myevic

Where is Ferox gone again )

messywrk, 2017-08-10:
I have installed your latest Build on my Predator - no problems so far..
Buttons/Screen/Vaping works fine!

messywrk, 2017-08-10:
BTW: Where do you get the Info which Box is compatible to what setting if you don't own the box yourself? I mean stuff like ( ISPRIMO1 || ISPRIMO2 || ISPREDATOR ) <- why is the predator similar to the primo1/2?
I'd love to help out for testing the Sinous P80 too, would that be comparable to ClockSelect's changes for the Primo Mini?

fOmey, 2017-08-10:
@messywrk have a read here I asked similar questions: #328

MicroSur, 2017-11-20:
SME (sur_myevic fork) is supported P80, but I don't know how to drive embedded led...