ClockSelect/myevic #332: Clockselect??? You're back???

Dil3mm4, 2017-07-25:
@ClockSelect  are you back brother????? 
If so, it's so nice to see you BACK!!!!

ClockSelect, 2017-07-26:
Well...

This week-end a friend of mines lent me a Primo and I have some hard time with the stock firmware so I relaunched my IDA... and since I was there, I fixed a few things.

To make it short, I have far less time than before to make this project live like it deserves and I'm sorry for that.

I'm really amazed to see all that has been done around it when I was away.

@Dil3mm4 I saw your doc yesterday on the MicroSur branch and I love it. Sure it's far better organized this way and I'll adopt it right away. Big thanks to you.

Dil3mm4, 2017-07-26:
I totally get your feelings. 

I've abandoned a few projects too for the same reason and feels bad, it's like not having time to "feed" your creature. 
But, as you can see, it's still strong. 
You inspired a lot of developers and I'm personally greateful to your dedication. 

When I found out this project (and the forks too), I really tried to understand your code to keep on, but I'm having hardtimes understanding the whole "thing" without somekind of documentation on routines, variables and so on.

At least there's some ppl better than me at it, that seems to fully understand what you've done so far. 

Hope you'll fine a bit more of time for it, I'd be really glad to help you out as far as I can, lending a hand to spread the software all-over, translating, reporting and understanding bugs (maybe applying fix, but guess that's too far for me :D).

Hope to see you and myevic rising stronger than ever. 

Thank you @clockselect .

tharealgc, 2017-07-28:
@ClockSelect 
Thank you for everything you have done so far man, deleting profiles? what? WOOOO!!!

Thank you, in fact thank you very much!

arafey, 2017-08-08:
Wow, you're actually back. Do you think you'll have time to work with some pull requests every once in a while? I, and I'm sure many others, would prefer to contribute to this valuable project while you're busy rather than fragment the community with forks (not against forking, but it's annoying when forks have desirable features/fixes but there isn't a single firmware that contains everything).

ClockSelect, 2017-08-17:
I don't really like working with pull requests, and I reject most of them. Reason is that most people wishing some feature just "patch" the code to have them working, and it's bad work. But still, It may give me ideas and I do not forbid such requests. If I feel it's woth implementing, I'll redo it the right way, kill the pull request, and commit the new feature. And we'll discuss it in pull request thread.