ClockSelect/myevic #337: Thoughts and suggestion

Dil3mm4, 2017-07-30:
Hey there!
It's almost a year that I'm switching/testing/enjoying various firmwares out there. 

And I'm taking a bit of time to write here the feature that I've recognized so far, that maybe, if @ClockSelect it's down with that, would be nice to see on myevic.

__TL;DR__ - There's no TLDR. Please read :D

### After X puffs in a row of X minutes the box turns off
As already done with a [commit](https://github.com/MicroSur/sur_myevic/commit/f24200305eecde13b4fe3791dfed296b78100d28) by Microsur, the box turns off after a certain number of puffs. 
In my opinion lacks a bit of customization, maybe letting the user choose, how long it's a session. 
It could be implemented a "session end" in minutes as a customizable parameter.

### Lack of documentation for Vaped ml/kJ feature 
That's purely a personal opinion. 
Reading the docs, I'm not able to define how to adjust the setting to my setup. 
It's stated that the default value runs on a 30W factory coil. 
So to setup a value on a 120W coil, the value should be multiplied per 4 (1440), if I understood well.
But the value has a limit on 1000. 
Dunno If I'm misreading or just to noob to understand it :D .
Any clarification would be nice.

### New Fire Button Behaviour
Would be nice to implement a X sec Fire on click as done by pharmaka in this [commit](https://github.com/pharmaka/tubomyevic/commit/413c8a036561cf429af4eb3579552e9dee44b868), letting the user choose:
* how many secs of "autofire";
* which behaviour they want to assign to the Fire Button (normal/timed);
* autofire interruption after clicking - button (or Fire again)

This should be helpful too, to get more statistics element for Vaped ml/kJ feature, increasing its accuracy.

### Re-read atom/coil function in clicks submenu 
I've enjoyed this feature a lot when rebuilding while I was on ArcticFox. 
Initializing a coil almost certainly highs up the ohm value and being able to redetect those settings with a click shortcut it's simply awesome.

### Possibility to re-assign 5 clicks on fire button
That's purely in the name of full customization. 
I'd like to choose between three options:
* None;
* Unlock/Lock Device; 
* Shutdown device;
 
I thought about ON/OFF settings only on 5 clicks to don't confuse the potential user that comes from OFW.

### After X minutes/seconds idle, the box locks itself
The title says it all.

### Full-fledged "always-ON" mode
Since the displays used in these boxes are mostly OLEDs, it would be cool to have an always on mode. 
Maybe with the locking feature described earlier, but with an always on display showing clock, day, that works as a lockscreen. 
It shouldn't impact the battery that much if on a reasonable contrast setup. 
(Or maybe if the "always-on" feature would be too much, just the lockscreen would be cool). 

### Software documentation
I know that @ClockSelect doesn't have a lot of time and maybe documenting the whole software it's boring as hell. 
But will help a lot developers like me, that doesn't get usually on this amazing low level when programming. 

Also if you're not willing (because of time or life dramas :D) to implement or consider this, please point out some reference in the source, maybe explaining where/how to implement it. 

Thank you for your time.

MicroSur, 2017-08-05:
When locking rez in TC, the real current resistance is locked. It's enough?

In my fork when battery  "it's not able to erogate the desidered watt" mod is in Low state. 
Other time it in full desired power. I think its better then power down when low batteries.
Weak battery notifying present as in stock fw.

Dil3mm4, 2017-08-05:
>When locking rez in TC, the real current resistance is locked. It's enough?`

I don't get your point here. 
Why you're pointing out the TC locking on live resistance? 
For this paragraph **"Re-read atom/coil function in clicks submenu**"? 

> In my fork when battery "it's not able to erogate the desidered watt" mod is in Low state.
Other time it in full desired power.

Let's make an example. 
At the moment, with the current behaviour, if you run on a wattage that your battery/ies it's not able to deliver, it returns "Battery Weak" and does not erogate power.
You have to "guess" which wattage the battery can erogate in VW with the power available in the battery. 
For the "average user" it's not that cool to receive that message with a 60-70% shown. 
That's why a behaviour like this would be better for average user:
```
if wattage > "WhatBatteryCanErogate" && VoltErogation > CutOff
erogate max wattage available
```
So it can understand at least which wattage the mod it's able to push out in that precise moment.
This wouldn't lead to auto power down.

MicroSur, 2017-08-05:
>Re-read atom/coil 
re-read to what point else? Cold rez of coil is important in TC only.

The mod does not know anything about the possibility of a battery.

Dil3mm4, 2017-08-05:
>The mod does not know anything about the possibility of a battery.

The box knows the current volt erogable by the current battery setup.
Otherwise it couldn't display the volt in main screen or in charge  screen neither calculate the percentage.
And wouldn't be able too, to calculate current wattage based on voltage.
Or I'm wrong?

The re-read atomizer/coil it's just a shortcut that would come handy when regenerating RDAs.
Burning coils always leads to an higher ohm than when on cold and instead of unscrewing the atom and screwing it again for a new reading, it would be useful to have a click shortcut to it.

MicroSur, 2017-08-05:
Wattage based on rez and output voltage u setup, not on battery voltage or its energy.

As I say, in my fork locking rez in TC is the same as u wish. Shortcut to it is lock procedure. 
In VW mode it is unnecessary.

Dil3mm4, 2017-08-05:
Do you use to lock coil before firing it up to reach its "real ohm" in TC? 
Still in VW, the coil reading in its static version would be unprecise.

MicroSur, 2017-08-05:
>Do you use to lock coil before firing it up to reach its "real ohm" in TC?

Yes, once or if it is necessary. Then cold rez is stored for use in TC. Stored rez not changed while fire.
More, U can manually correct rez in coil management menu.

Dil3mm4, 2017-08-05:
>U can manually correct rez in coil management menu.
Exactly. The shortcut was the mean to avoid manual correction in my idea.

MicroSur, 2017-08-05:
What rez u wish to have when re-read coil? Cold, hot or some else?
How the mod can determine your wish?

Dil3mm4, 2017-08-05:
Live. 
If you're resyncing your rez, that's supposed to be the live one.

MicroSur, 2017-08-05:
Live one may be not cold rez and u have to press re-read again and again. Correction will be easy way.
U want to do this in VW too? What for?

Dil3mm4, 2017-08-05:
Just for a visual correct value, nothing else.

Dil3mm4, 2017-08-05:
By the way, there's nothing else in the list that seems good for you? :)

MicroSur, 2017-08-06:
I do not want autofire (dangerous :) ), Unlock/Lock Device (switch on/off better), "session end" in minutes (I made this for stock FW with patches but never used)
I have clock in Off state (with fire click).
May be auto switch off timer and re-read rez in VW like I do for TC, from edit mode...

Dil3mm4, 2017-08-06:
> I do not want autofire (dangerous :) )

Dangerous if you puff for >4 sec. 
Otherwise it won't cause any harm to the user nor to the DC converter. 

> "session end" in minutes (I made this for stock FW with patches but never used)
The thing about the session end is that currently I don't know how the puffreset() acts. 
If it resets after X minutes or what.

MicroSur, 2017-08-06:
By the way, mod re-read coil every time and correct on screen automatically in VW. 
We need to have Locked_VW_res, like in TC for this feature better...

>If it resets after X minutes or what.

It is new counter variable for puff time, like for puffs count... Counter goes down to zero, switch mod off, then set up again...

Dil3mm4, 2017-08-06:
>It is new counter variable for puff time, like for puffs count... Counter goes down to zero, switch mod off, then set up again...

Is the counter based on screen sleep?

MicroSur, 2017-08-06:
It base on what u want. I made it based on puff time.

Dil3mm4, 2017-08-06:
Yes but when the counters reset?? 
Based on what?

enricorov, 2017-08-23:
Just bumping this thread, I second some of the feedback.

I'd like to have a timeout feature, since I always forget to lock the box, and the autofiring feature. Both because I'm lazy, that's for sure.