ClockSelect/myevic #338: Wismec firmware update program not recognizing my Rx2/3 after using custom firmware

Angelshover, 2017-08-20:
I added the myvic custom firmware a few weeks ago to my rx2/3. Its has world pretty good since. However I recently have been having a problem where the device won't let me go passed 72.3watts in power mode. I don't know if I changed a setting or something so I figured I'd just restore it to stock firmware and than redo it. 

Problem is my computer nor the official Wismec firmware updating program will recognize my mod. It charges but still say "none device" and doesn't let me update.

After searching around here I found someone say changing this setting expert->USB->HID* so I tried all of those options and none worked. 

Anyone ever have this issue or have a possible solution. Thanks in advance.

SEEKNDESTROY, 2017-08-20:
Yesterday I was tryig to do a fresh install of this custom firmware but,
wasn't able to upgrade my RX200S to the latest stock firmware... I kept
getting the device not found message. However, I was able to install an
older stock firmware version with the installer that came in the zip file.
Strange thing, I got the same message when I tried to install the newest
version after installing the older version... not a big deal since I'm
happy with ClockSelect's firmware.

Try installing an older version and make sure to use the installer included
with that version. Here is the link
http://www.wismec.com/software/wismec-tc-software-for-upgrading-reuleaux-rx23/

Dil3mm4, 2017-08-21:
@SEEKNDESTROY you could try to use Safe Mode to flash your device. 
Remove batteries. 
Press Plus button while connecting the device. 
Flash it. :)

ClockSelect, 2017-08-30:
Even if your device is theorically capaple of outputting 200W or so, it will not do so under any conditions. Depending on the resistance of the coil, the power will be limited by the 9 volts or 50 Amps cap, and the available power will be limited accordingly.