ClockSelect/myevic #339: Info on splash screen

enricorov, 2017-08-23:
I'm sure it's been posted before, but I couldn't find anything on the topic. 

I've been using openevic for a while, and liked the splash screen showed when powering on the box. I remember reading that feature had been implemented before the fork. How do I include my own?

MicroSur, 2017-08-26:
>How do I include my own?
With FW Editors. My VTCFont or NFE.

enricorov, 2017-08-28:
Thanks a lot, I'll do that.

closed 2017-08-28 by enricorov

enricorov, 2017-08-28:
For all those having my same problem:

- Download the latest evic.bin file
- Get VTCFont, and load the .bin
- The splash screen in question is at location FF, scroll the left bar until you get there
- Edit it as you please
- When done, press 'Save all'
- Flash the .bin file with your tool of choice